ros2/rcutils #4: concat should let user decide the memory allocation strategy

mikaelarguedas, 2017-03-28:
User should pass allocated strings and concat should not make the choice of calling malloc for the user. e.g. realtime users may have this memory preallocated at startup and don't want spurious malloc in their realtime loop: https://github.com/ros2/c_utilities/blob/bb2fd057f15c9556f602bc2da192aab61cdd39f3/src/concat.c#L40

Karsten1987, 2017-03-28:
I'll assign this to me and try to get this done in the near future.

mikaelarguedas, 2017-03-28:
I can handle it if you want, I'm shoveling some c_utilities stuff these days.
If you take care of it, can you also make the delimiter argument optional and default to NULL, and add docblocks for these functions ? Thanks!

Karsten1987, 2017-06-14:
i close this in favor of using `rcutils_format_string` instead.
https://github.com/ros2/rcutils/blob/master/include/rcutils/format_string.h

closed 2017-06-14 by Karsten1987