ros2/rcutils #6: pass custom allocator struct to utilities function

Karsten1987, 2017-03-29:
all functions should be somehow capable of handling custom memory allocators in the style of RCL.
https://github.com/ros2/rcl/blob/master/rcl/include/rcl/allocator.h

mikaelarguedas, 2017-04-17:
@ros2/team to review the associated PRs

dirk-thomas, 2017-04-20:
What is the status of this ticket? All referenced PRs are closed or merged.

mikaelarguedas, 2017-04-26:
@Karsten1987 bump

Karsten1987, 2017-04-26:
split and concat still have native malloc calls which have to be addressed in the future. So this ticket stays open for now.

Karsten1987, 2017-06-07:
linux: [image]
osx: [image]
windows: [image]

dirk-thomas, 2017-06-13:
This needs a new round of CI imo.

wjwwood, 2017-06-14:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

mikaelarguedas, 2017-06-14:
The windows failure is unrelated so I think this is good to go

mikaelarguedas, 2017-06-14:
Note: the new warning on Windows is also unrelated

wjwwood, 2017-06-14:
This is still failing because some `rcl` test uses `rcutils_string_array_fini()` and it wasn't updated. I'll look into it tomorrow, as I need this to open my pr.

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

wjwwood, 2017-06-14:
Looks like @Karsten1987 fixed those missing changes and ran ci. Thanks! Those new prs need review, and I'll have to collect the ci jobs when I get to the office.

wjwwood, 2017-06-14:
CI @Karsten1987 ran:

- Linux:
  - [image]
- Linux-aarch64:
  - [image]
- macOS:
  - [image]
- Windows:
  - [image]

wjwwood, 2017-06-14:
Windows failure is unstable for unrelated reasons, I don't think Linux-aarch64 is required since Linux passed. I'll go ahead and merge.

Karsten1987, 2017-06-14:
i'll close this ticket and refer to this ros2/rcl#127 for the lifecycle changes.

closed 2017-06-14 by Karsten1987