ros2/rcutils #7: C flags when not building tests

codebot, 2017-04-01:
I ran across this while building c_utilities on gcc 4.9.2:
```
[ 12%] Building C object CMakeFiles/c_utilities.dir/src/cmdline_parser.c.o
/home/chip/sros2/src/ros2/c_utilities/src/cmdline_parser.c: In function ‘cli_option_exist’:
/home/chip/sros2/src/ros2/c_utilities/src/cmdline_parser.c:22:3: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
   for (size_t i = 0; i < (size_t)(end - begin); ++i) {
   ^
/home/chip/sros2/src/ros2/c_utilities/src/cmdline_parser.c:22:3: note: use option -std=c99, -std=gnu99, -std=c11 or -std=gnu11 to compile your code
CMakeFiles/c_utilities.dir/build.make:62: recipe for target 'CMakeFiles/c_utilities.dir/src/cmdline_parser.c.o' failed
make[2]: *** [CMakeFiles/c_utilities.dir/src/cmdline_parser.c.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/c_utilities.dir/all' failed
make[1]: *** [CMakeFiles/c_utilities.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```
I think this is because the line specifying C11 is here:
https://github.com/ros2/c_utilities/blob/master/CMakeLists.txt#L35
which is only executed when tests are being built. If a build is kicked off without tests, it fails (or at least it did for me). Moving the `CMAKE_C_FLAGS` and `CMAKE_CXX_FLAGS` definitions up in the file (outside the `if (BUILD_TESTING)` block) allowed the build to work for me.

mikaelarguedas, 2017-04-02:
that is a good point, I tried to address this on [the compile_options branch](https://github.com/ros2/c_utilities/commit/db24473a0d67bc6f063bb6cca3c2524fbf159829) but didn't have time to test it or open a PR accordingy.

Although MSVC doesn't support it IIRC so all c code declare loop variable beforehand, very surprised that this passed on windows in the first place o_O. So unless MSVC is now c99 compliant :tada: , I think that the c code should also be modified to declare `size_t i;` before the for loop.

wjwwood, 2017-04-02:
Yeah, since MSVC doesn't support C11 I don't think we should try and turn it on. We don't need it really anyways. As @mikaelarguedas said, I think the code should be updated to declare the `i` outside the loop.

codebot, 2017-04-02:
implemented in #8

closed 2017-04-02 by codebot

dirk-thomas, 2017-04-03:
The code itself (not the tests) should be compiled with all warnings enabled. Currently that is still not the case (only when tests are enabled).