ros2/rcutils #13: types should be namespaced with utilities_

wjwwood, 2017-04-12:
Currently types like `string_array_t` could easily, imo, collide with any arbitrary other system. I believe it would be better to prefix these types with `utilities_`.

I can foresee that this will be annoying to do, but in that case I'd argue we should try for a shorter and simpler package name/prefix name instead of avoiding namespacing in some places.

Karsten1987, 2017-04-12:
tadaaa :)
https://github.com/ros2/rmw/pull/100

wjwwood, 2017-04-12:
Awesome thanks!

I have some other thoughts on this string array interface as well though. I guess some of them are dependent on https://github.com/ros2/c_utilities/issues/6. For example, what happens if `calloc` returns `NULL` here:

https://github.com/ros2/c_utilities/blob/31f11ddf987b5f585bcd77494948cd1c8f664fba/src/string_array.c#L37

Karsten1987, 2017-04-12:
yup, I am on it. This will all go somehow together with #6 and https://github.com/ros2/rmw/issues/94

dirk-thomas, 2017-04-12:
I understand and agree with the "need"for namespacing. But why use `utilities_` instead of `c_utilities_`? For all other code the namespacing follows the package name (even in ROS 1).

wjwwood, 2017-04-12:
I was just following the style of the rest of the package, which uses just utilities as the prefix. I agree and I've also pointed out that c_utilities is not very unique, nor is it a convenient suffix, but I didn't have an alternative name in mind.

Karsten1987, 2017-04-13:
How do we proceed here? Are we pushing these changes forward until we come up with a better name and then do a global refactoring?

dirk-thomas, 2017-04-13:
I think we should make a decision on the name now. If we update all pending PRs immediately or in another round doesn't matter.

wjwwood, 2017-04-14:
I created an issue to follow up on the naming issue: https://github.com/ros2/c_utilities/issues/17

Karsten1987, 2017-04-14:
closing this with https://github.com/ros2/rmw/pull/100

closed 2017-04-14 by Karsten1987

wjwwood, 2017-04-14:
Thanks @Karsten1987!