ros2/rcutils #17: change name of this repository

wjwwood, 2017-04-14:
Some people, including myself, feel that this repository/package name is not ideal and that we should change it, see: https://github.com/ros2/c_utilities/issues/13#issuecomment-293560955 (and perhaps other conversations). If you disagree please chime in here.

When/if we select a new name, we should act quickly to update all instances to minimize impact on git history and cross-references. Also, we should update this package to use the full package name in the manual namespacing throughout.

I talked with @Karsten1987 offline about this and I came up with two criteria which could improve the name of this package:

- A shorter name
  - since this is package contains C code, its functions and types need to be manually namespaced by prefixing them with the package name
  - @dirk-thomas pointed out that some shortcuts have been taken, using `utitlities_` instead of the full `c_utilities_` for example
    - I agree with him, that using the exact package name in the namespace is a very useful convention
  - a shorter name will keep it convenient and prevent the desire for shortcuts like these
- A name with some reference to our project or organization
  - My concern is that `c_utilities_` and especially `utilities_` is not a very unique name
  - a unique name helps avoid symbol collisions and makes it more obvious that some part of the code is part of our project
    - for example, others thought we were adding an external dep: https://github.com/ros2/rmw/pull/100#issuecomment-293644996
    - Other examples of this are [`apr` (Apache Portable Runtime)](https://apr.apache.org/) and [`glib` GLib](https://developer.gnome.org/glib/2.26/) which is part of Gnome.
      - both are widely reused despite containing a reference to a project or organization in the name

Based on these goals I just threw out some alternatives, but I'm not really attached to any of them so please others chime in with options:

- `osrfcutils` **OSRF C Util**itie**s**
  - could be extended for other languages, e.g. for C++ as `osrfcpputils`
- `osrfcu` **OSRF C U**tilities
  - could be extended for other languages, e.g. for C++ as `osrfcppu`
- `osrflibc` **OSRF Lib**rary for **C**
  - could be extended for other languages, e.g. for C++ as `osrflibcpp`
- `libosrfc` **Lib**rary used at **OSRF** for **C**
  - could be extended for other languages, e.g. for C++ as `libosrfcpp`
  - library names could be modified so that they come out as `libosrfc.ext` rather than `liblibosrfc.ext`
- `osrfc` **OSRF C** utilities library
  - could be extended for other languages, e.g. for C++ as `osrfcpp`
  - library names would naturally be `libosrfc.ext`
- `rcutils` **ROS C Util**itie**s**
  - could be extended for other languages, e.g. for C++ as `rcpputils`

There were some others, but I didn't really like them or they collided with existing things, for example anything containing `roslib` might be confused with:

https://github.com/ros/ros/tree/lunar-devel/core/roslib

Or that `ros2*` sounds like you're converting to something from ros, or anything containing `comm` and not knowing if that's short for "common" or "communication".

My personal favorite atm is probably `osrfc` or `libosrfc`, as I think `utils` or `utilities` is a bit redundant or catch all, sort of like misc or common. To me leaving off words like that implies the same thing. I also think that they're pretty unlikely to collide with anything else and are relatively short.

dirk-thomas, 2017-04-14:
I personally don't like the idea of having `osrf` in the name. Just imagine if that package would have been created a few years ago - different company name. They same might be the case in the future.

Therefore I am in favor of `rcutils`. It encode the important pieces imo: ROS, C, utils. And leaves room for similar packages for different languages.

Regarding the content I just want to mention that I think we need to be careful what to put inside that package. While we mainly want to avoid duplication we should be very careful not to put too many things into that package. It should aim to provide "utilities" which makes downstream code easier / less verbose (that is what e.g. `apr` and `glib` are about). But we should try to avoid providing crucial concepts which are then becoming must haves in downstream code. (I know this sounds fuzzy and is difficult to describe but I wanted to mention it anyway.)

codebot, 2017-04-14:
+1 rcutils 👍

<https://github.com/ros2/c_utilities/issues/17#issuecomment-294079347>, or mute
the thread

clalancette, 2017-04-14:
I'd personally suggest something more like: roscutils.  It's a little longer, but then it is very clear that this is for ROS.

Karsten1987, 2017-04-14:
👍  `rcutils`

IanTheEngineer, 2017-04-14:
:+1: to `rcutils` as well, or something equivalently `ROS-y`. I agree with avoiding `osrf` in the namespace. I am constantly haunted by the `hlr` namespace in our Rethink code. Organization names change from time to time, but the project-code tends to live on :)

tfoote, 2017-04-14:
I'd take the other side and say that this is not ROS specific. We are using it to support the development of ROS2.0 but it certainly could be reused outside of ROS. It's true that it likely won't reach the scope of `osrf_pycommon`. 

Following the goal of making downstream code less verbose, and following the examples of `apr` and `glib` I'd suggest shortening it to `rcu` This would come from it's core purpose as ROS C Utilities however like `apr` it can have a life of it's own if others want to pick it up. It's also short so usages are clear. And it's matches nicely with rmw and rcl for length making readability easier too.

IanTheEngineer, 2017-04-14:
Well put, Tully. Changing my +1 to `rcu`.

wjwwood, 2017-04-14:
Thanks for all the feedback so far everyone.

---

@clalancette 

> I'd personally suggest something more like: roscutils. It's a little longer, but then it is very clear that this is for ROS.

The preference seems to be for `rcutils` do you feel strongly about `ros` versus just `r` for the suffix, or are you okay with the shorter hand?

---

@tfoote

> It's true that it likely won't reach the scope of `osrf_pycommon`.

For me, `osrf_pycommon` came in under different constraints, because it was being used by catkin and ament, so there wasn't a clear "default" user, so the organization made the most sense at the time. To be honest, I'm not 100% happy with the name of that thing either, but it's the best I could come up with at the time. Also I consider `osrf_pycommon` to be "an OSRF common python library", not "the OSRF common python library", so if another project at OSRF needed common Python stuff they could make another library, but the hope is that they wouldn't need to do so. I feel the same about this package, it certainly won't be the only utilities library in ROS at some point given the breadth of people adding stuff to the ecosystem, but is **a** utility package used by ROS.

I understand the arguments against an institution name, but I guess I just don't mind if the company name changes or something like that in the future. It's part of the history regardless of the name and is in the copyright, etc. It's still unique and wouldn't be too hard for someone to figure out why `osrf` is in the name. That being said, I like the concise nature of the `rc*` names, and I'm perfectly fine with that.

> I'd suggest shortening it to `rcu`

I like the shorter name, but my only hesitation is that `rcl*` stands for ROS Client Library, whereas this would be `rcu` ROS C Utilities, so the `c` stands for different things in each name, which might be confusing. I don't think that mix up would cause an issue, but maybe just misconceptions. I don't feel strongly about it, but that's why I didn't suggest it earlier.

codebot, 2017-04-14:
Awesome, I also change my +1 to rcu  😬

fewer characters is a good thing. since it's 2017 we could also remove all
vowels :)  but rcu sounds great.

mikaelarguedas, 2017-04-14:
:+1: for rcutils. agreed with the potential confusion of `rcu` that @wjwwood highlighted in his previous comment

wjwwood, 2017-04-14:
Ok, I've added a summary of the preferences (ranked choice where implied) to the original post:

https://github.com/ros2/c_utilities/issues/17#issue-221713329

If you'd like to add your preference or update it please comment and I'll do it. If you want to edit the post yourself, you can but make sure to refresh right before editing to avoid overwriting each other.

Please register your opinion by end of day Saturday, so I can start to put the change in motion after that.

IanTheEngineer, 2017-04-15:
> "I'd suggest shortening it to rcu"
> I like the shorter name, but my only hesitation is that rcl* stands for ROS Client Library, whereas this would be rcu ROS C Utilities, so the c stands for [...]

I think the `c` in `rcu` is can be interpreted interchangeably between "client" and "c-language" in this context. All ros2 language clients that depend on on the `rcl` will depend on this c-library as well, making `rcu` quite fitting.

wjwwood, 2017-04-17:
Looks like there is a slight preference for `rcu`. I'll go forward with that.

wjwwood, 2017-04-17:
tl;dr we'll use `rcutils` unless another issue comes up.

@clalancette thanks for that insight. I searched for `"rcu" library`, but I missed `uruc` amongst the credit unions and university websites. Given that I, I'll retract my support for `rcu` but what I support instead doesn't matter I think, because no matter what I believe `rcutils` wins. Luckily I had not started because my other prs that need to be merged first took longer to clean up than I had hoped yesterday.

wjwwood, 2017-04-20:
I'll leave this open until the repository is renamed and the ros2.repos file is update.

closed 2017-04-20 by wjwwood