ros2/rcutils #27: log throttle tests are failing on macOS

wjwwood, 2017-06-06:
See: http://ci.ros2.org/job/nightly_osx_release/481/testReport/

These changes were merged yesterday: https://github.com/ros2/rcutils/pull/23

I don't have an example of these tests passing on the above link nightly (this is the first night), so I can't say if it is flakey or not. I assume so since I assume there was a passing CI job for macOS before merging. I don't see the same failure on Linux or Windows atm.

dirk-thomas, 2017-06-06:
This was already mentioned yesterday: https://github.com/ros2/rclcpp/pull/333#issuecomment-306316157

See #28 for longer timeouts.

closed 2017-06-06 by dirk-thomas

wjwwood, 2017-06-07:
macOS release is back to green: http://ci.ros2.org/view/nightly/job/nightly_osx_release/482/

PR #28 (2017-06-06): increase test timeouts for OS X
Not that a single run says much... [image]

Fixes #27.