ros2/rcutils #32: rcutils logging macro warning

wjwwood, 2017-06-15:
I was using the new logging macros like this:

```c
#include "rcutils/logging_macros.h"
// ...
    RCUTILS_LOG_ERROR(
      "failed to clean up on error (leaking memory): '%s'", rcutils_get_error_string_safe());
```

But I get a warning like this:

```
In file included from /Users/william/ros2_ws/src/ros2/rcutils/src/split.c:26:
/Users/william/ros2_ws/build/rcutils/include/rcutils/logging_macros.h:948:13: warning: token pasting of ',' and __VA_ARGS__ is a GNU extension [-Wgnu-zero-variadic-macro-arguments]
    format, ## __VA_ARGS__)
            ^
/Users/william/ros2_ws/build/rcutils/include/rcutils/logging_macros.h:93:68: warning: token pasting of ',' and __VA_ARGS__ is a GNU extension [-Wgnu-zero-variadic-macro-arguments]
      rcl_log(&__rcutils_logging_location, severity, name, format, ## __VA_ARGS__); \
                                                                   ^
2 warnings generated.
```

I looked into it briefly, but I didn't dig too deep. Am I doing something wrong or is this just a warning we should fix in the logging macros?

The warning only references the line on which the header is included, but if I remove all uses of the logging macro (but leave the include) the warning goes away.

clalancette, 2017-06-15:
This page has a pretty good explanation of the GNU extensions that is probably causing the problem: https://gcc.gnu.org/onlinedocs/cpp/Variadic-Macros.html .  It seems to me that we should probably try to remove the __VA_ARG__ token pasting from the macros and see how that does (it will definitely work in gcc, I'm less sure about clang and even less sure about MSVC).

clalancette, 2017-06-15:
FYI, I did a quick branch that removes the ## from __VA_ARGS__ in the logging macros, and artificially put a print somewhere in rcl.  The resulting CI is here, we'll see whether we get warnings or not:

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

mikaelarguedas, 2017-06-15:
Unfortunately this fix forces users to have a variadic argument making it impossible to do
`RCUTILS_LOG_ERROR("I have no argument");` so we either need a way to handle empty va_args list or use a different logic on a per platform basis.
* Linux [image]

```
In file included from /home/mikael/work/ros2/ros2_ws/src/ros2/rcl/rcl/src/rcl/common.c:26:0:
/home/mikael/work/ros2/ros2_ws/src/ros2/rcl/rcl/src/rcl/common.c: In function ‘rcl_impl_getenv’:
/home/mikael/work/ros2/ros2_ws/install_debug_isolated/rcutils/include/rcutils/logging_macros.h:93:79: error: expected expression before ‘)’ token
       rcl_log(&__rcutils_logging_location, severity, name, format, __VA_ARGS__); \
                                                                               ^
/home/mikael/work/ros2/ros2_ws/install_debug_isolated/rcutils/include/rcutils/logging_macros.h:945:3: note: in expansion of macro ‘RCUTILS_LOG_COND_NAMED’
   RCUTILS_LOG_COND_NAMED( \
   ^
/home/mikael/work/ros2/ros2_ws/src/ros2/rcl/rcl/src/rcl/common.c:37:3: note: in expansion of macro ‘RCUTILS_LOG_ERROR’
   RCUTILS_LOG_ERROR("No variadic arg");
```

clalancette, 2017-06-15:
Oh, interesting.  That page I linked to above led me to believe that that would work, but I guess I misunderstood.  We'll have to come up with something else.

dirk-thomas, 2017-06-15:
I was able to reproduce it. It seems to only happen with C code. The C++ tests are fine. I will add a C test for this and come up with a fix...

@clalancette Can you test branch be deleted?

clalancette, 2017-06-15:
Yeah, that's fine, I'll delete it.

dirk-thomas, 2017-06-15:
This is a good case which would be resolved if we would use `set(CMAKE_C_STANDARD 11)`.

dirk-thomas, 2017-06-15:
Switched to C11 in 05ab6c091c396e88b04e937123814a2e889a8fe9.

PR #35 (2017-06-16): add c test using logging macros
Fixes #32.

For now the PR only adds a test which reproduced the problem:

* Linux [image]
* OS X [image]

Looking into a fix...
closed 2017-06-16 by dirk-thomas