ros2/rcutils #56: rcutils logging cleanup

dhood, 2017-10-06:
I am thinking about how to specify logger severities progammatically, and would like input regarding memory de-allocation.

Suppose that we already have an appropriate data structure that represents the mapping of logger names to severities, and users have a call that let's them specify severities of loggers at runtime. These are the options that I see for managing the memory of that data structure:
1. We allow the memory for the severity data structure to grow at runtime.
1. We pre-allocate an amount of memory for the data structure and don't let usage go above that amount.
1. [Alternative for completeness] Don't store the data structure; just check a config file or other data source each time.

I've been assuming that (1) is preferred, but that might not be the case. If (1) is preferred, and if the memory allocation is managed in C, we will also have to have a way to de-allocate the memory, but there is no symmetric concept of the `rcutils_logging_initialize` that de-allocates memory at the moment.

These are the options that I see if we want to continue with (1):
1. If sticking to C, create a fini function that de-allocates the severity data structure. It will be logging users' responsibility to call it when done with the logging system.
    1. Optionally the function could be called from within `rcl_shutdown` to assist ROS users. However, they might still want to call logging macros after `rcl_shutdown`.
1. Switch to C++ underneath and use automatic memory management.
1. Automatic memory management in C (I've read that we could implement/pull in a garbage collector).

Implementing the fini for option (1) is not necessarily an issue, but I'm not sure that requiring users to call it is quite reasonable, which is why I'm stopping to ask if we should go in the C or C++ direction.
Perhaps it's possible to have an implementation somewhere in the middle, where C++ users don't have to call the fini, but C users do: e.g. the memory management implemented in C, and optionally if C++ support is built we have a smart pointer that automatically calls the C fini on destruction.

What do others think regarding the two points?

wjwwood, 2017-10-06:
I think the best default option is to let it grow at runtime (option 1). If that's a deal break in the future we can have a separate function that sets a limit, then we can preallocate and refuse to exceed it. Just make sure the function that would normally increase the capacity can fail (needs to anyway if allocating memory).

Unless I misunderstand it, then should only happen at the beginning of the program (for the most part) and so it shouldn't be a concern for real-time or anything (we just tell them to add all the loggers and set the severity at the beginning).

As for deallocation I would say we have a function that can deallocate the memory, but we don't require anyone to call it. If the program is terminated, it doesn't need to clean up, but if someone wanted to clean it up (in order to initialize it again) they can do so with this function. If you really want to clean it up, you could use the library unloading event to do so (since after unloading the library where the global data is it will be unavailable anyways), see: http://en.cppreference.com/w/c/program/atexit

dhood, 2017-10-06:
Thanks @wjwwood, I'd been looking at `rcutils_shutdown` thinking it was a requirement that it was called explicitly, but from your comment I realised that it's not necessary that users explicitly call the cleanup function.

Then I should be fine to go ahead and allocate memory when necessary, and will provide the cleanup function for users to use optionally. Thanks!

closed 2017-10-06 by dhood