ros2/rcutils #58: find() and find_last() functions return ambiguous value

clalancette, 2017-10-20:
If a NULL or empty string is passed into `rcutils_find()` or `rcutils_find_last()`, they return 0.  However, it may also be the case that the delimiter being searched for is in the zero'th position, and the caller doesn't have any way to disambiguate this.  As discussed in https://github.com/ros2/rcutils/pull/57#pullrequestreview-69972479, we should probably change this so that `rcutils` and `rcutils_find_last` return something different; either a negative value (which would require us to change the return type), or the size of the string (indicating it wasn't found).  The latter will obviously only work on empty strings, not NULL arguments.  In either case, we need to audit all of the callers and make sure they do the correct thing.

mikaelarguedas, 2017-11-07:
For completeness, @sloretz [suggested](https://github.com/ros2/rcutils/pull/57#discussion_r145823592) to return `SIZE_MAX` that would allow to not change the return type and should work in the various mentioned scenarios.

mikaelarguedas, 2017-11-27:
@clalancette FYI: moving this in review

PR #74 (2017-11-27): Update the rcutils_find* APIs to return SIZE_MAX on invalid arguments.

fixes #58 

CI:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]
closed 2017-11-27 by clalancette