ros2/rcutils #70: Force flush for rcutils logging

dhood, 2017-11-15:
We have code that forces flushing of stdout in some places, particularly so that launch file prints sync. This includes python scripts [that call](https://github.com/ros2/demos/blob/1208b81650c0e8bad5ad4db300f93ef65ecc027f/topic_monitor/topic_monitor/scripts/data_publisher.py#L107) `sys.stdout.flush()`. After switching to use rcutils logging (via python wrappers), `sys.stdout.flush()` will do nothing for python scripts, and they will not have a way to force flush.

Consider an environment variable similar to https://github.com/ros/ros_comm/pull/1012.

dhood, 2017-11-15:
Python scripts can use `PYTHONUNBUFFERED` for now, so this is less of a priority. This will still be useful for the C/C++ code to have forced flushing configured externally as opposed [to hard-coded](https://github.com/ros2/demos/blob/8354081d55c627b60b8e1229bdcd84d6ec036068/lifecycle/src/lifecycle_talker.cpp#L256), and to have control over all languages with one configuration.

yenWu, 2018-03-01:
Hello @dhood. I don't see there will be an elegant way rather than ugly hack for all languages. Maybe we can do something in shell command. However, buffering mechanism is specified by POSIX. Have anyone got any idea about that? I see the label here, and wanna try it out.

dhood, 2018-03-03:
@yenWu thanks for your interest!

The objective is to remove the need for hard-coding anything to force flushing in ROS applications themselves (as you noted it changes for each language). This can be accomplished with an environment variable instead. That way, the decision to use or not use unbuffered output can be made when the process is _run_. The good thing is that `rclpy` and `rclcpp` both use `rcutils` for logging, and so the environment variable will affect both, without needing a different approach for all languages.

https://github.com/ros/ros_comm/pull/1012 shows how this has been accomplished in ROS 1. It would be good to copy that PR into `rcutils`, so that we call `fflush` after [console output is printed](https://github.com/ros2/rcutils/blob/c6e63774f09664e78fd442720b1289c194a7213a/src/logging.c#L558). Note that instead of `get_environment_variable` you can use `rcutils_get_env` [like we do for other logging initialisation](https://github.com/ros2/rcutils/blob/c6e63774f09664e78fd442720b1289c194a7213a/src/logging.c#L80).

How does that sound?

dirk-thomas, 2018-03-28:
Addressed by #98.

closed 2018-03-28 by dirk-thomas