ros2/rcutils #77: Rename logger "severity threshold" to "level"

dhood, 2017-12-01:
[Please speak up if you have any objections, otherwise I'll go ahead with this after lunch.]

Each log message has a particular severity level e.g. DEBUG. Loggers can be configured to only process messages at and above a level e.g. INFO by default, currently with the "severity threshold" of the logger.

While writing logging demos I've formed the opinion that using "severity threshold" to specify this in function names is long winded.

I propose to rename functions such as [`rcutils_logging_set_logger_severity_threshold`](https://github.com/ros2/rcutils/blob/77a7e9e3cca139a74e44df1e5755c7e3c03cf1aa/include/rcutils/logging.h#L293) to `rcutils_logging_set_logger_level`, and update wrapping in rclpy to match.

This will match terminology in [ros1](https://github.com/ros/ros_comm/blob/4383f8fad9550836137077ed1a7120e5d3e745de/tools/rosconsole/include/ros/console.h#L88), python logging module and log4cxx.

mikaelarguedas, 2017-12-02:
* Linux [image] (linter failure unrelated and fixed in https://github.com/ros2/rosidl/commit/69956510f5a57500b2f51fc14484c0ff35060b8d)
* Linux-aarch64: no ARM job so far
* macOS [image] (pending executor)
* Windows [image]  (linter failure unrelated and fixed in https://github.com/ros2/rosidl/commit/69956510f5a57500b2f51fc14484c0ff35060b8d)

dirk-thomas, 2017-12-03:
As far as I understand in log4j the two terms `level` and `threshold` are used in a very specific way. `Loggers` are using `level` since they describe one specific level (e.g. `info`). On the other hand `Appenders` are using `threshold` since they identify all levels equal and above a specific level.

Therefore I would argue the term "threshold" makes more sense in cases where it is being used to describe which log event are being handled and which are being ignored.

wjwwood, 2017-12-03:
I'd say that "threshold" has just been dropped here, the words that were exchanged in my opinion are "level" and "severity". For instance, "log level threshold" makes total sense to me, and the most explicit for me would be "log severity level threshold".

In all cases, I think it's clear enough what's going on with the current name, but I'm agnostic to changing it again.

dhood, 2017-12-03:
I agree that, functionally, loggers apply a threshold to severity levels. But even log4j refers to this as `Logger.setLevel` for loggers as opposed to explicitly stating in the function name that it changes the severity threshold. Users know that setting the logger level causes messages with severity levels at and above `level` to be processed. Yes, including threshold would be more explicit, but as I mentioned it was clunky to use and diverges from other logging library APIs, hence the change.

closed 2017-12-04 by dhood