ros2/rcutils #78: RCUTILS_SAFE_FWRITE_TO_STDERR with variable length messages

dhood, 2017-12-01:
If you pass a message of type `const char *` to `RCUTILS_SAFE_FWRITE_TO_STDERR` e.g. https://github.com/ros2/rcutils/blob/954ae9bb071436ad281f31c388deaf39afe0e465/src/split.c#L116

then the `sizeof(msg)` calculation in the macro gives the size of the pointer:
https://github.com/ros2/rcutils/blob/954ae9bb071436ad281f31c388deaf39afe0e465/include/rcutils/error_handling.h#L44

From what I can tell the `count` passed to `fwrite` is not a _max_ size, but _the_ size, so I think this line can cause buffer overrun if the msg is less than length 8. That doesn't happen with our existing code because they're all longer (instead our messages get truncated), but it could happen.

What's the appropriate fix here?
- continue using `sizeof(msg)` and only use this macro with static buffers
- switch to `strlen(msg)` and only use this macro will null-terminated buffers (also a risk)

clalancette, 2017-12-01:
Ouch.  I don't think we can stick with `sizeof(msg)` here; there is just way too much risk that someone will pass an allocated pointer to it (and there isn't really a good way to enforce that restriction at compile-time).  While `strlen(msg)` is always a risk, it is a risk you have to take when using C, so I'd suggest we go with that.

wjwwood, 2017-12-01:
You can use `strnlen_s` with a relatively large bound to prevent a really malicious case:

http://en.cppreference.com/w/c/string/byte/strlen

clalancette, 2017-12-01:
> You can use strnlen_s with a relatively large bound to prevent a really malicious case:

Oh, nice, I didn't know about that.  That sounds like a good thing to use.

dhood, 2017-12-01:
To double check my understanding: `strnlen_s` with a large number doesn't remove the chance of buffer overrun, it's just that overrun length will have an upper bound, yeah?

wjwwood, 2017-12-01:
Yeah.

PR #80 (2017-12-03): Use strlen / strnlen_s in RCUTILS_SAFE_FWRITE_TO_STDERR
fixes #78 
Otherwise it does not work correctly for dynamic buffers

CI forthcoming, at which point I'll put this into review once I'm sure it works cross-platform
closed 2017-12-03 by dhood