ros2/rcutils #88: sanity check for rcutils_steady_time_now

serge-nikulin, 2018-02-05:
`rcutils_steady_time_now` currently trusts the OS specific implementations to provide truly monotonic time values. Issue #86 and [Internet](https://stackoverflow.com/questions/3657289/linux-clock-gettimeclock-monotonic-strange-non-monotonic-behavior#3657385) show the monotonicity claim should be verified.

mikaelarguedas, 2018-02-07:
Thanks @serge-nikulin, we were indeed using "true" monotonic clocks in the past and changes the behavior to match what we get from chrono at the c++ level. linking to the relevant issue here for book-keeping: https://github.com/ros2/rcutils/issues/43

serge-nikulin, 2018-02-07:
@mikaelarguedas, because of non-monotonic Windows steady timer (due to overflow in previous ROS2 implementation, fixed by https://github.com/ros2/rcutils/pull/87) I had a silent failure in a publisher's executor.

This fix forces the system to non-silent error state in similar failure mode on all platforms.

clalancette, 2022-04-22:
I'm going to close this as sort of a duplicate of #43.  If you disagree, feel free to reopen (but I think we need an explicit actionable thing to do here).

closed 2022-04-22 by clalancette