ros2/rcutils #142: redundancy: De-duplicate `find_library` code from RMW libs, move here?

EricCousineau-TRI, 2019-03-25:
If I follow the build-from-source instructions here:
https://index.ros.org/doc/ros2/Installation/Linux-Development-Setup/#get-ros-2-0-code

I notice that the `find_library` code is duplicated 3 times:
```
$ find src/ -name '*.h' -o -name '*.c' -o -name '*.hpp' -o -name '*.cpp' | xargs grep 'find_library'
src/ros2/rosidl_typesupport/rosidl_typesupport_cpp/src/type_support_dispatch.hpp:std::string find_library_path(const std::string & library_name);
src/ros2/rosidl_typesupport/rosidl_typesupport_cpp/src/type_support_dispatch.hpp:        std::string library_path = find_library_path(library_name);
src/ros2/rosidl_typesupport/rosidl_typesupport_cpp/src/type_support_dispatch.cpp:std::string find_library_path(const std::string & library_name)
src/ros2/rosidl_typesupport/rosidl_typesupport_c/src/type_support_dispatch.hpp:std::string find_library_path(const std::string & library_name);
src/ros2/rosidl_typesupport/rosidl_typesupport_c/src/type_support_dispatch.hpp:        std::string library_path = find_library_path(library_name);
src/ros2/rosidl_typesupport/rosidl_typesupport_c/src/type_support_dispatch.cpp:std::string find_library_path(const std::string & library_name)
src/ros2/rmw_implementation/rmw_implementation/src/functions.cpp:std::string find_library_path(const std::string & library_name)
src/ros2/rmw_implementation/rmw_implementation/src/functions.cpp:    std::string library_path = find_library_path(env_var);
```

It would be really nice to de-duplicate this code. If possible, could this be hoisted into this library? (I'm guessing re-written in C?)

From perusing this code, it looks there are current conditionals that depend on `_WIN32` and `__APPLE__`, so it seems appropriate?

This effectively comes from: https://raw.githubusercontent.com/ros2/ros2/release-crystal-20190314/ros2.repos
Source: https://github.com/ros2/ros2/blob/8d34d6550a77dcfe4828466e53f0cc5401be77fe/ros2.repos

(Came about while tinkering with consuming libs from Bazel (https://index.ros.org/doc/ros2/Installation/Linux-Development-Setup/#get-ros-2-0-code).)

dirk-thomas, 2019-03-25:
The code could stay C++ and be moved into [rcpputils](https://github.com/ros2/rcpputils).

EricCousineau-TRI, 2019-03-25:
Sounds good! Do you want me to migrate this issue there, or we can keep this one?

I'd be happy to do the legwork and submit some prototype PRs.