ros2/rcutils #159: Cross compilation rcutils with Android NDK

roderick-koehle, 2019-05-14:
Cross compiling rcutils with Clang/Android NDK (Android NDK / 19.2.5345600) 
results in following compilation error:

```
Scanning dependencies of target rcutils
[  4%] Building C object CMakeFiles/rcutils.dir/src/process.c.o
/home/EU/dekoehlr/work/ros2_core_ws/src/ros2/rcutils/src/process.c:53:26: error: use of undeclared identifier 'program_invocation_name'
  const char * appname = program_invocation_name;
                         ^
1 error generated.
CMakeFiles/rcutils.dir/build.make:332: recipe for target 'CMakeFiles/rcutils.dir/src/process.c.o' failed
make[2]: *** [CMakeFiles/rcutils.dir/src/process.c.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/rcutils.dir/all' failed
make[1]: *** [CMakeFiles/rcutils.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
```

As possible fix for this issue is to check for the  `__clang__` flag and use `getprogname()` instead of `program_invocation_name` specific to GCC/glibc.

```diff --git a/src/process.c b/src/process.c
index 08122de..fd77318 100644
--- a/src/process.c
+++ b/src/process.c
@@ -47,7 +47,7 @@ char * rcutils_get_executable_name(rcutils_allocator_t allocator)
   RCUTILS_CHECK_ALLOCATOR_WITH_MSG(
     &allocator, "invalid allocator", return NULL);
 
-#if defined __APPLE__
+#if defined __APPLE__ || __clang__
   const char * appname = getprogname();
 #elif defined __GNUC__
   const char * appname = program_invocation_name;
```

The script to reproduce the results is as follows:
```bash
#!/usr/env bash
#
# ROS2 Cross compiler build setup for Android SDK
#
# Prerequisites: 
# - installed Android NDK and cmake build tool addons.

export WORK_DIR=$HOME/work
export ROS2_WORKSPACE=$WORK_DIR/ros2_core_ws

# Provide Android SDK location
export JAVA_HOME=$HOME/local/android-studio/jre
export ANDROID_SDK=$HOME/local/android-sdk
export ANDROID_NDK=$HOME/local/android-sdk/ndk-bundle

export PATH=$ANDROID_SDK/tools/bin:${PATH}
export PATH=$ANDROID_SDK/platform-tools:${PATH}

# Android build toolchain
export ANDROID_ABI=armeabi-v7a
export ANDROID_NATIVE_API_LEVEL=android-21
export ANDROID_TOOLCHAIN_NAME=arm-linux-androideabi-clang
export ANDROID_STL=c++_shared

# Download and install ROS2 sources
mkdir -p $ROS2_WORKSPACE
cd $ROS2_WORKSPACE
if [ ! -d src ]; then
  mkdir src
  wget -nc https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos
  vcs import src < ros2.repos

  # Patch rcutils/src/process.c to support compilation with clang
  echo -e "50c50\n< #if defined __APPLE__\n---\n> #if defined __APPLE__ || __clang__" | patch src/ros2/rcutils/src/process.c
fi

# Partition packages into host and device build
# Note: Amend index is dependency of rclcpp_components - do not build for host.
AMENT_LINT=$(echo ament_{lint_common,lint_auto,lint_cmake,flake8,pep257,copyright})
AMENT_CMAKE=$(echo ament_cmake ament_cmake_{core,test,gmock,gtest,pytest,libraries,nose,python,pytest})
AMENT_INDEX="ament_index_python"

PKG_AMENT="${AMENT_CMAKE} ${AMENT_LINT} ${AMENT_INDEX} ament_cmake_ros ament_package ament_pclint ament_clang_format osrf_pycommon uncrustify_vendor osrf_testing_tools_cpp"
PKG_PYTHON="python_cmake_module rosidl_generator_py rclpy rosidl_runtime_py launch launch_ros launch_testing_ament_cmake"
PKG_EXTRAS="intra_process_demo orocos_kdl rclpy resource_retriever tf2_ros libcurl_vendor tinyxml2_vendor tinyxml_vendor"
PKG_CONNEXT="rosidl_typesupport_connext_c rosidl_typesupport_connext_cpp rmw_connext_cpp rmw_connext_shared_cpp" 
PKG_OPENSPLICE="rosidl_typesupport_opensplice_c rmw_opensplice_cpp rosidl_typesupport_opensplice_cpp"

# Ignore packages which cannot be build on target device
PKG_IGNORE="rcl_logging_log4cxx  ${PKG_PYTHON} ${PKG_EXTRAS} ${PKG_CONNEXT} ${PKG_OPENSPLICE} ${PKG_AMENT}"
PKG_EXAMPLES=$(echo examples_rclcpp_minimal_{action_client,action_server,client,composition,publisher,subscriber,service,timer})
PKG_TARGET="${PKG_EXAMPLES} ament_index_cpp composition demo_nodes_cpp_native lifecycle logging_demo"

echo "* Host build of Ament:"
colcon build --symlink-install \
  --build-base build/host \
  --install-base install/host \
  --cmake-clean-cache \
  --packages-skip-build \
  --packages-up-to ${PKG_AMENT}

source install/host/local_setup.bash
  
echo "* Device build of ROS2:"
colcon build --symlink-install  \
  --build-base build/device \
  --install-base install/device \
  --packages-up-to ${PKG_TARGET} \
  --packages-ignore ${PKG_IGNORE} \
  --cmake-clean-cache \
  --cmake-args \
    --no-warn-unused-cli \
    -DCMAKE_BUILD_TYPE=Release \
    -DCMAKE_TOOLCHAIN_FILE=${ANDROID_NDK}/build/cmake/android.toolchain.cmake \
    -DANDROID_NATIVE_API_LEVEL=${ANDROID_NATIVE_API_LEVEL} \
    -DANDROID_TOOLCHAIN_NAME=${ANDROID_TOOLCHAIN_NAME} \
    -DANDROID_STL=${ANDROID_STL} \
    -DANDROID_ABI=${ANDROID_ABI} \
    -DANDROID_NDK=${ANDROID_NDK} \
    -DTHIRDPARTY=ON \
    -DCOMPILE_EXAMPLES=OFF \
    -DBUILD_TESTING=OFF \
    -DCMAKE_FIND_ROOT_PATH="$ROS2_WORKSPACE/install/host;$ROS2_WORKSPACE/install/device" \
    -DRCL_LOGGING_IMPLEMENTATION=rcl_logging_noop
```

Besides this minor issue, I was able to cross compile the core components of ROS2 on Android. Keep up the good work!

Best regards,

Roderick

clalancette, 2019-05-14:
Thanks for the bug report.

> As possible fix for this issue is to check for the `__clang__` flag and use `getprogname()` instead of `program_invocation_name` specific to GCC/glibc.

That change seems reasonable to me.  If you don't mind opening a PR, I'd be happy to review it.

roderick-koehle, 2019-05-24:
Looking at the Android Bionic Headers, the BSD style `getprogname` is introduced in `__ANDROID_API__>=21`.

If I search for getprogname in the NDK, the definition of getprogname
'ndk-bundle/toolchains/llvm/prebuilt/linux-x86_64/sysroot/usr/include/stdlib.h'

```c
/* BSD compatibility. */

#if __ANDROID_API__ >= 21
const char* getprogname(void) __INTRODUCED_IN(21);
void setprogname(const char* __name) __INTRODUCED_IN(21);
#endif /* __ANDROID_API__ >= 21 */
```
in the public repository of bionic, the header is slightly different to the installation,
https://android.googlesource.com/platform/bionic/+/refs/heads/ndk-release-r19/libc/include/stdlib.h

My suggestion is to use `__APPLE__ || (__ANDROID__ && __ANDROID_API__>=21)` as a feature test macro to test for BSD compatibility of stdlib.

In NDK releases prior to <=11, both clang and gcc toochains where available. Above check is valid only for the clang toolchain and does not support other BSD variants.
https://developer.android.com/ndk/downloads/revision_history

clalancette, 2019-05-24:
> My suggestion is to use `__APPLE__ || (__ANDROID__ && __ANDROID_API__>=21)` as a feature test macro to test for BSD compatibility of stdlib.

That sounds good to me.  Thanks for the feedback, will update the PR accordingly.

PR #162 (2019-05-29): getprogname() is the correct API to use on Android.
While we are here, also fix the __CYGWIN__ macro, which was
missing the final two underscores.

This should fix #159 , though I haven't been able to test that myself.  @roderick-koehle , this is a different solution than you proposed, but I think is more accurate.  Please give it a test and let me know if it fixes the issue for you.
closed 2019-05-29 by clalancette