ros2/rcutils #165: Add environment variable to configure logging level

windelbouwman, 2019-06-18:
Today I tried to debug my ROS2 setup, but I was unable to get detailed logging output. It would be good to be able to set the default log level via an environment variable.

My idea would be to be able to do this:

- export RCUTILS_LOG_LEVEL=DEBUG
- run executable with more debug output than normally the case.

Is this a good idea? Or did I miss some other option to troubleshoot my setup?

clalancette, 2019-06-18:
There are several ways to increase the logging levels in ROS 2.  There is a [concept page](https://index.ros.org/doc/ros2/Concepts/Logging/) and a [tutorial page](https://index.ros.org/doc/ros2/Tutorials/Logging-and-logger-configuration/) .

I'm going to close this out, but if you feel that those aren't sufficient, feel free to keep commenting, reopen, or ideally provide a PR to https://github.com/ros2/ros2_documentation to improve the documentation.

closed 2019-06-18 by clalancette

ruffsl, 2021-03-16:
@clalancette , per the link cited, is there a ticket tracking external configuration of loggers at runtime?

> In the future there will be a generalized approach to external configuration of loggers at runtime (similar to how rqt_logger_level in ROS 1 allows logger configuration via remote procedural calls). This concept is not yet officially supported in ROS 2. In the meantime, this demo provides an example service that can be called externally to request configuration of logger levels for known names of loggers in the process.  
> https://docs.ros.org/en/foxy/Tutorials/Logging-and-logger-configuration.html#logger-level-configuration-externally

clalancette, 2021-03-17:
> @clalancette , per the link cited, is there a ticket tracking external configuration of loggers at runtime?

There are quite a few open bugs in this area:

* https://github.com/ros2/launch/issues/488
* https://github.com/ros2/rclcpp/issues/978
* https://github.com/ros2/launch/issues/383
* https://github.com/ros2/ros2/issues/425
* https://github.com/ros2/rosbag2/issues/288
* https://github.com/ros2/rcl_logging/issues/6

mlanting, 2021-04-16:
Is anyone currently working on these? If not, I'll take a look at them.

clalancette, 2021-04-16:
> Is anyone currently working on these? If not, I'll take a look at them.

Not that I know of.

If you are interested in this, we'd welcome the help.  In that case, I'll suggest starting with https://github.com/ros2/design/issues/314 , and coming up with a design document that describes what we want the logging subsystem to be.  From there, we can determine what pieces we have and what is missing.  With all of that information in hand, we can then move forward with implementing the remaining pieces.

mlanting, 2021-04-28:
I'll take a look at this then and try to get a first-pass PR up in the next week.

mlanting, 2021-05-06:
PR is up here: https://github.com/ros2/design/pull/315