ros2/rcutils #168: console output should be line-buffered by default

rotu, 2019-07-13:
Right now, info-level logging output is not line-buffered by default, whereas error-level logging output is. This is a problem for usability, since it causes extremely confusing logs when an error message is flushed before an info message. The buffering of stdout, while appropriate for some Linux programs, is inappropriate when diagnostic logging output is emitted from both streams stderr and stdout.

The consequences of having `RCUTILS_CONSOLE_STDOUT_LINE_BUFFERED=1` when it should be 0 is a tiny (if noticeable) performance hit. The consequences of having it 0 when it should be 1 is incomprehensibly misordered logging output, often with info messages not emitted until a node dies. Because of this, unset `RCUTILS_CONSOLE_STDOUT_LINE_BUFFERED` should be treated as if it had the value 1, not 0.

rotu, 2019-07-14:
@mjeronimo @sagniknitr @dirk-thomas 
I strongly think this is the safer, saner default. Please object if you think this is wrongheaded or there is a better solution.

rotu, 2019-07-14:
An example of bad logging in a complex, multi-node launch. It's very hard to determine the order things happen in between multiple nodes. If line flushing, it would be clear that `dwb_controller` is created *before* `amcl` terminates.

```
ros2 launch openrover_demo nav2.launch.py
[INFO] [launch]: All log files can be found below /home/snuc/.ros/log/2019-06-10-17-13-22-508548-rhea-30016
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [lifecycle_manager-1]: process started with pid [30046]
[INFO] [map_server-2]: process started with pid [30047]
[INFO] [amcl-3]: process started with pid [30048]
[INFO] [world_model-4]: process started with pid [30052]
[INFO] [navfn_planner-5]: process started with pid [30054]
[INFO] [motion_primitives_node-6]: process started with pid [30063]
[INFO] [bt_navigator-7]: process started with pid [30071]
[INFO] [dwb_controller-8]: process started with pid [30078]
[lifecycle_manager-1] [WARN] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[amcl-3] [WARN] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[navfn_planner-5] [WARN] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[world_model-4] [WARN] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[dwb_controller-8] [WARN] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[amcl-3] [ERROR] [rclcpp]: failed to create wait set: the given context is not valid, either rcl_init() was not called or rcl_shutdown() was called., at /opt/ros/master/src/ros2/rcl/rcl/src/rcl/wait.c:130
[amcl-3] terminate called after throwing an instance of 'std::runtime_error'
[amcl-3]   what():  Failed to create wait set in Executor constructor
[INFO] [lifecycle_manager-1]: process has finished cleanly [pid 30046]
[ERROR] [amcl-3]: process has died [pid 30048, exit code -6, cmd '/home/snuc/ros2_ws/install/lib/nav2_amcl/amcl __node:=amcl __params:=/home/snuc/ros2_ws/install/share/openrover_demo/config/nav2.yaml'].
[INFO] [motion_primitives_node-6]: process has finished cleanly [pid 30063]
[INFO] [bt_navigator-7]: process has finished cleanly [pid 30071]
[lifecycle_manager-1] [INFO] [lifecycle_manager]: Creating
[lifecycle_manager-1] [INFO] [rclcpp]: signal_handler(signal_value=2)
[lifecycle_manager-1] [INFO] [lifecycle_manager]: Destroying
[motion_primitives_node-6] [INFO] [motion_primitives]: Configuring Spin
[motion_primitives_node-6] [INFO] [motion_primitives]: Robot: Creating
[motion_primitives_node-6] [INFO] [motion_primitives]: Configuring BackUp
[motion_primitives_node-6] [INFO] [motion_primitives]: Robot: Creating
[motion_primitives_node-6] [INFO] [rclcpp]: signal_handler(signal_value=2)
[motion_primitives_node-6] [INFO] [motion_primitives]: Robot: Destroying
[motion_primitives_node-6] [INFO] [motion_primitives]: Robot: Destroying
[bt_navigator-7] [INFO] [bt_navigator]: Creating
[bt_navigator-7] [INFO] [rclcpp]: signal_handler(signal_value=2)
[bt_navigator-7] [INFO] [bt_navigator]: Destroying
[navfn_planner-5] [INFO] [navfn_planner]: Creating
[navfn_planner-5] [INFO] [rclcpp]: signal_handler(signal_value=2)
[navfn_planner-5] [INFO] [navfn_planner]: Destroying
[INFO] [navfn_planner-5]: process has finished cleanly [pid 30054]
[map_server-2] [INFO] [map_server]: Creating
[map_server-2] [INFO] [rclcpp]: signal_handler(signal_value=2)
[map_server-2] [INFO] [map_server]: Destroying
[world_model-4] [INFO] [world_model]: Creating
[world_model-4] [INFO] [global_costmap.world_model]: Creating
[world_model-4] [INFO] [rclcpp]: signal_handler(signal_value=2)
[world_model-4] [INFO] [world_model]: Destroying
[world_model-4] [INFO] [global_costmap.world_model]: Destroying
[INFO] [map_server-2]: process has finished cleanly [pid 30047]
[INFO] [world_model-4]: process has finished cleanly [pid 30052]
[dwb_controller-8] [INFO] [dwb_controller]: Creating
[dwb_controller-8] [INFO] [local_costmap.dwb_controller]: Creating
[dwb_controller-8] [INFO] [rclcpp]: signal_handler(signal_value=2)
[dwb_controller-8] [INFO] [dwb_controller]: Destroying
[dwb_controller-8] [INFO] [local_costmap.dwb_controller]: Destroying
[INFO] [dwb_controller-8]: process has finished cleanly [pid 30078]
```

mjeronimo, 2019-07-15:
@rotu No objections here; I think it should be line buffered by default. I was mystified at first when this change was first introduced and the output was not line buffered. It took me a bit to figure out what was going on. Definitely a usability issue and an unexpected surprise for someone new.

rotu, 2019-07-16:
There are a couple possible solutions here. To wit:

1. Make *all* diagnostic output (including `RCLCPP_INFO` and `RCLCPP_DEBUG`) go to `stderr`. This frees up `stdout` for program-defined usage and is probably the cleanest and most idiomatic solution.
2. Set `RCUTILS_CONSOLE_STDOUT_LINE_BUFFERED` to 1 by default. This is probably the minimal solution that solves the problem in all cases.
3. Make *all* diagnostic info go to a custom stream buffer that is flushed whenever an error is logged and/or at a frequent program-defined interval and/or whenever the logger's node spins.

ivanpauno, 2019-07-18:
@rotu https://github.com/ros2/launch/pull/265 should have solved the problem.
Can you double check if it solved your problem?

ivanpauno, 2019-07-18:
> @rotu [ros2/launch#265](https://github.com/ros2/launch/pull/265) should have solved the problem.
> Can you check if it solved your problem?

Well, it has finally been reverted, because it caused some failures on CI.

rotu, 2019-07-18:
thanks, @ivanpauno. I can see that the patch does make launch behave a bit better by default. The root cause here, though is not in launch. It's that we use two different streams for diagnostic output, for which RC_UTILS_CONSOLE_STDOUT_LINE_BUFFERED is just a partial workaround, and so is changing the behavior of launch. The more I think about it, the more I think logging shouldn't use stdout at all.

ivanpauno, 2019-07-18:
> thanks, @ivanpauno. I can see that the patch does make launch behave a bit better by default. The root cause here, though is not in launch. It's that we use two different streams for diagnostic [...]

IMO, that's not standard.
When you redirect stdout to a pipe, the standard behavior is that it's block buffered (that happens in any process).

In some cases forcing line buffering (or not buffering at all) and avoiding fully buffering is nice, and for that reason you can use the environment variable. But I don't think that forcing it as a default is a good idea. Providing an environment variable for this is usual, for example python provides `PYTHONUNBUFFERED` environment variable for avoiding it.

Logging everything to `stderr` is also quite unusual. If not, redirecting `stdout` and `stderr` to different files would be impossible.

----

I think that `launch` should handle this problem without the user doing something special. That will be the case after `emulate tty` PR gets merged.

rotu, 2019-07-18:
>  When you redirect stdout to a pipe, the standard behavior is that it's block buffered (that happens in any process).
The performance issues only come into play when you have (1) massive amounts of data and (2) stdout is being directly written to disk. If you have enough logs to notice the performance, you are logging too much. [Here are the actual performance numbers I measured](https://github.com/ros2/rcutils/pull/169#issuecomment-511572942).

> Logging everything to stderr is also quite unusual.

It seems like it's the standard, actually.  In particular, note that the stream usage is by its intended output, not its *severity*:
https://unix.stackexchange.com/questions/331611/do-progress-reports-logging-information-belong-on-stderr-or-stdout

>     Variable: FILE * stdout
>     The standard output stream, which is used for normal output from the program.
> 
>     Variable: FILE * stderr
>     The standard error stream, which is used for error messages and diagnostics issued by the program.

Stdout is usually used for the production of a program, such the found lines in `grep` or the list of files in `ls`. This doesn't exist for most ROS programs - the output goes over the network to other nodes or to hardware devices.

wjwwood, 2019-07-19:
> The performance issues only come into play when you have (1) massive amounts of data and (2) stdout is being directly written to disk. If you have enough logs to notice the performance, you are logging too much. Here are the actual performance numbers I measured.

Correct me if I'm wrong, but I don't think we can change the blocking behavior selectively, we'd need to do that for the entire process or not at all. In which we'd be constraining the utility of all programs that happen to use ROS logging in any way, e.g. something like `ros2 topic echo --csv /points > some_file.csv` comes to mind.

> Stdout is usually used for the production of a program, such the found lines in grep or the list of files in ls. This doesn't exist for most ROS programs - the output goes over the network to other nodes or to hardware devices.

I'm not sure I agree, much of the log output from ROS programs is what it is doing, some are long running and others a short running, but I don't feel that it's fundamentally different just because they typically also send data over the network.

I don't think of a ROS program as being any different than say, `git`. Just spot checking, `git` uses both stdout and stderr:

```
william@kisra ~/ros2_ws/src/ros2/launch (git:revert-276-revert-265-emulate_tty:12c0563)
% git st
On branch revert-276-revert-265-emulate_tty
Your branch is up to date with 'origin/revert-276-revert-265-emulate_tty'.

nothing to commit, working tree clean

william@kisra ~/ros2_ws/src/ros2/launch (git:revert-276-revert-265-emulate_tty:12c0563)
% git st > /dev/null

william@kisra ~/ros2_ws/src/ros2/launch (git:revert-276-revert-265-emulate_tty:12c0563)
% cd

william@kisra ~
% git st
fatal: not a git repository (or any of the parent directories): .git

william@kisra ~
% git st > /dev/null
fatal: not a git repository (or any of the parent directories): .git
```

Logging anything to stderr that's not an error or warning is weird in my opinion, and so I'd severity does play a role.

----

I think matching the behavior of other languages and frameworks by doing nothing by default but having an environment variable to change the blocking behavior (a la `PYTHONUNBUFFERED`) makes the most sense.

Also, I can see cases where you'd like even the info and debug log messages to go to stderr as well, leaving stdout for application usage (though I'm not convinced that should be the default), so having programmatic and maybe environmental ways to change that behavior would be good too.

rotu, 2019-07-19:
> Correct me if I'm wrong, but I don't think we can change the blocking behavior selectively, we'd need to do that for the entire process or not at all. In which we'd be constraining the utility of [...]

You're 100% correct. Also, the output of ros2 topic does not use the logging macros since its output is not logging info. `ros2 topic echo --csv` should have all its product be csv, and debug / info messages coming out over stdout breaks that contract.

> I don't think of a ROS program as being any different than say, `git`. Just spot checking, `git` uses both stdout and stderr

The reason `git status` logs its fatal output to the command line is because there is no current repo, so it can't report a meaningful status. (it's the same behavior as, say, `git remote` on a non-repo folder).

The first hit for "debug" in the `git` codebase shows that debug output goes out over stderr:
https://github.com/git/git/blob/b7bd9486b055c3f967a870311e704e3bb0654e4f/mailmap.c#L8

> Also, I can see cases where you'd like even the info and debug log messages to go to stderr as well

Agreed it would be best for logging to be configurable and extensible. There are cases where it might make sense for other debug hooks, like sending debug info out over, e.g., UART or other such craziness.

rotu, 2019-07-20:
As another point of reference, `python3` logging comes out `stderr`:

```
➜ python3 -c "import logging; logging.basicConfig(level=logging.INFO); logger = logging.getLogger(__name__); logger.info('test')" > /dev/null 
INFO:__main__:test
```

mjeronimo, 2019-07-22:
While the typical output of a ROS node is via its messages, there is nothing to prevent a node from creating useful, non-diagnostic, output on stdout. I haven't really seen nodes do this - generate output that is then further piped to another ROS node/process for additional processing - but it's certainly possible. So, from my perspective, all of the logging output from a ROS node is diagnostic, no matter the severity level (INFO, WARNING, ERROR) and would be appropriate for stderr, leaving stdout for the application/node to use. Imagine trying to write a UNIX-style console application that could be piped to others to generate a command chain. Unfortunately, the library you're using to implement this command is generating output on stdout. The library would be encroaching into the application's domain, IMO. So, in the ROS case, I'd say the node should own stdout and the library (ROS) should output its diagnostic info to stderr.

commit a3f04b1b (2020-02-10): Always log on stderr

This follows the convention for human-readable diagnostic output on Linux systems.

fix #168

Make the stream configurable.

We default the stream to stderr, but allow the user to change
it to stdout by setting RCUTILS_CONSOLE_LINE_BUFFERED to 1.
commit 5c882db7 (2020-02-10): Always log on stderr

This follows the convention for human-readable diagnostic output on Linux systems.

fix #168

Make the stream configurable.

We default the stream to stderr, but allow the user to change
it to stdout by setting RCUTILS_CONSOLE_LINE_BUFFERED to 1.
commit d3538f7a (2020-02-12): Always log on stderr

This follows the convention for human-readable diagnostic output on Linux systems.

fix #168

Make the stream configurable.

We default the stream to stderr, but allow the user to change
it to stdout by setting RCUTILS_CONSOLE_LINE_BUFFERED to 1.
commit f62f2e0c (2020-02-12): Always log on stderr

This follows the convention for human-readable diagnostic output on Linux systems.

fix #168

Make the stream configurable.

We default the stream to stderr, but allow the user to change
it to stdout by setting RCUTILS_CONSOLE_LINE_BUFFERED to 1.
closed 2020-02-28 by clalancette