ros2/rcutils #170: Potential buffer overflow

danielwangksu, 2019-07-22:
Function call `rcutils_find(str + i, '}')` can returns 0, if a malformed function.
if so `size_t token_len = chars_to_end_delim - 1;` will cause token_len =  18446744073709551615 and cause overflow
https://github.com/ros2/rcutils/blob/2497ff5e28f749a6c55d060d55faf438bf224e64/src/logging.c#L634-L648

hidmic, 2019-08-08:
@danielwangksu for `rcutils_find(str + i, token_end_delimiter);` to return `0`, a `}` must be found at `str + i`. But before that, a `{` is searched for. Therefore `str + i` will be at a `{` by the time it looks for a `}`, and `chars_to_end_delim > 1` awalys holds. 

Are you proposing to `assert` on it in case there's a logical error in this code or `rcutil_find`s?

danielwangksu, 2019-08-08:
@hidmic thank you for the feedback! It was discovered by coverity. I think we can mark it as false alarm then. thanks.

closed 2019-08-13 by danielwangksu