ros2/rcutils #174: Provide compile-time ROS2 version information

serge-nikulin, 2019-08-12:
As ROS2 matures and introduces new API, we need ways to conditionally compile code against different, possibly compile-time incompatible, versions of ROS2.

Common ways to do it are:
- Define version macros or constants in C or C++ header file, e.g. (from FDRS code):
    ```c
    #define FASTCDR_VERSION_MAJOR 1
    #define FASTCDR_VERSION_MINOR 0
    #define FASTCDR_VERSION_MICRO 9
    #define FASTCDR_VERSION_STR "1.0.9"
    ```
- Define version information in a [CMake file](https://gitlab.kitware.com/cmake/community/wikis/doc/tutorials/How-to-create-a-ProjectConfig.cmake-file) and later generate a header file with C/C++ version macros or constants.

dirk-thomas, 2019-08-12:
That sounds like a good idea. Please consider to contribute pull requests for packages where you see the need for it.

serge-nikulin, 2019-08-12:
@dirk-thomas,

I can do a naive versioninig: plain C/C++ macros that should be upgraded manually before each build.

A "grownup", a right way to do versioning is to keep the last built version in some external DB and increment the minor version automatically and major/minor versions in a semi-auto way (via Jenkins commands?). This approach IMO should be implemented by ROS2 insiders who have full access to build infrastructure.

@dirk-thomas, should I go ahead with the naive approach pull request?

dirk-thomas, 2019-08-12:
Each package already has the version number declared in the `package.xml` file which also gets bumped for every release. That value is available in CMake as `<pkgname>_VERSION` after the manifest file was parsed (that happens either in `ament_package()` or when `ament_package_xml()` is called any time earlier). So that information could be used for the macros which will need to be generated using `configure_file()`.

serge-nikulin, 2019-08-12:
@dirk-thomas, see #175

aprotyas, 2022-04-29:
@clalancette how broad is this issue? rclcpp already generates a version header. Is there also a need for that in rcutils?

clalancette, 2022-04-29:
> @clalancette how broad is this issue? rclcpp already generates a version header. Is there also a need for that in rcutils?

I have no idea, honestly.  Maybe @serge-nikulin can weigh in.

serge-nikulin, 2022-05-07:
@aprotyas 
> > @clalancette how broad is this issue? rclcpp already generates a version header. Is there also a need for that in rcutils?

This issue is very old. Back in 2019 ROS2 did not have any version information.
You can close this issue I guess.

Back then I wanted to generate version information for every generated library.
Rationale: in many shops, all files must have version information, and many safety and security standards require version tagging of every distributed file.

clalancette, 2022-05-07:
Based on the previous comment, closing.  Feel free to reopen if you change your mind and would like to continue.

closed 2022-05-07 by clalancette