ros2/rcutils #189: Default logging format design

SteveMacenski, 2019-11-13:
Was there a design discussion around the default logging format defined here (https://github.com/ros2/rcutils/blob/ff8ca5803d32ee7f91cd50c99cf36741f104c681/src/logging.c#L65)?

I'd think the default you'd want a unix timestamp to determine relative timing of when events or logs occur. I can submit a PR to change this is folks agree. I can say from experience working with Navigation2 that this has been a pain point, which I know can be worked around setting `RCUTILS_CONSOLE_OUTPUT_FORMAT` however I'd think a timestamp is a reasonable default behavior.

fujitatomoya, 2019-11-14:
@SteveMacenski 

i totally do agree with you.

SteveMacenski, 2019-11-25:
#190

closed 2019-11-25 by SteveMacenski