ros2/rcutils #204: test_logging_output_format tests broken

rotu, 2020-02-05:
The tests in `test_logging_output_format.py` tests give worrying results in the CI log and when run locally. It seems like it may be a failure in the launch framework or in this package's usage of it.

Note particularly the error, which I think is saying that the test framework is sending a signal to a subprocess before it is set up to receive it: "RuntimeError: Signal event received before subprocess transport available."

http://build.ros2.org/job/Eci__nightly-cyclonedds_ubuntu_bionic_amd64/175/consoleText

```
5: Test command: /usr/bin/python3 "-u" "/tmp/ws/install_isolated/ament_cmake_test/share/ament_cmake_test/cmake/run_test.py" "/tmp/ws/test_results/rcutils/test_logging_long_messages.xunit.xml" "--package-name" "rcutils" "--output-file" "/tmp/ws/build_isolated/rcutils/launch_test/CHANGEME.txt" "--command" "/usr/bin/python3" "-m" "launch_testing.launch_test" "/tmp/ws/src/ros2/rcutils/test/test_logging_long_messages.py" "--junit-xml=/tmp/ws/test_results/rcutils/test_logging_long_messages.xunit.xml" "--package-name=rcutils"
5: Test timeout computed to be: 10
5: -- run_test.py: invoking following command in '/tmp/ws/build_isolated/rcutils':
5:  - /usr/bin/python3 -m launch_testing.launch_test /tmp/ws/src/ros2/rcutils/test/test_logging_long_messages.py --junit-xml=/tmp/ws/test_results/rcutils/test_logging_long_messages.xunit.xml --package-name=rcutils
5: [INFO] [launch]: All log files can be found below /home/buildfarm/.ros/log/2020-02-03-01-38-09-940584-47eaa799d93c-15284
5: [INFO] [launch]: Default logging verbosity is set to INFO
5: test_logging_output (rcutils.TestLoggingLongMessages)
5: Test executable output against expectation. ... [INFO] [test_logging_long_messages-1]: process started with pid [15288]
5: [test_logging_long_messages-1] [INFO] [name1]: XxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxX (func() at file:42)
5: [test_logging_long_messages-1] [INFO] [name2]: X42xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxX (func() at file:42)
5: ok
5: 
5: ----------------------------------------------------------------------
5: Ran 1 test in 0.005s
5: 
5: OK
5: [INFO] [test_logging_long_messages-1]: process has finished cleanly [pid 15288]
5: 
5: ----------------------------------------------------------------------
5: Ran 0 tests in 0.000s
5: 
5: OK
5: -- run_test.py: return code 0
5: -- run_test.py: verify result file '/tmp/ws/test_results/rcutils/test_logging_long_messages.xunit.xml'
 5/36 Test  #5: test_logging_long_messages .......   Passed    0.35 sec
test 6
      Start  6: test_logging_output_format

6: Test command: /usr/bin/python3 "-u" "/tmp/ws/install_isolated/ament_cmake_test/share/ament_cmake_test/cmake/run_test.py" "/tmp/ws/test_results/rcutils/test_logging_output_format.xunit.xml" "--package-name" "rcutils" "--output-file" "/tmp/ws/build_isolated/rcutils/launch_test/CHANGEME.txt" "--command" "/usr/bin/python3" "-m" "launch_testing.launch_test" "/tmp/ws/src/ros2/rcutils/test/test_logging_output_format.py" "--junit-xml=/tmp/ws/test_results/rcutils/test_logging_output_format.xunit.xml" "--package-name=rcutils"
6: Test timeout computed to be: 10
6: -- run_test.py: invoking following command in '/tmp/ws/build_isolated/rcutils':
6:  - /usr/bin/python3 -m launch_testing.launch_test /tmp/ws/src/ros2/rcutils/test/test_logging_output_format.py --junit-xml=/tmp/ws/test_results/rcutils/test_logging_output_format.xunit.xml --package-name=rcutils
6: [INFO] [launch]: All log files can be found below /home/buildfarm/.ros/log/2020-02-03-01-38-10-295070-47eaa799d93c-15290
6: [INFO] [launch]: Default logging verbosity is set to INFO
6: 
6: ----------------------------------------------------------------------
6: Ran 0 tests in 0.000s
6: 
6: OK
6: Task exception was never retrieved
6: future: <Task finished coro=<LaunchService._process_one_event() done, defined at /tmp/ws/install_isolated/launch/lib/python3.6/site-packages/launch/launch_service.py:293> exception=RuntimeError('Signal event received before subprocess transport available.',)>
6: Traceback (most recent call last):
6:   File "/tmp/ws/install_isolated/launch/lib/python3.6/site-packages/launch/launch_service.py", line 295, in _process_one_event
6:     await self.__process_event(next_event)
6:   File "/tmp/ws/install_isolated/launch/lib/python3.6/site-packages/launch/launch_service.py", line 315, in __process_event
6:     visit_all_entities_and_collect_futures(entity, self.__context))
[... cut ...]
6: [test_logging_output_timestamps-4] '1580722690.309443789' '1580722690309443789'
6: [test_logging_output_timestamps-4] '1580722690.309470948' '1580722690309470948'
6: test_logging_output (rcutils.TestLoggingOutputFormatAfterShutdown)
6: Test all executables output against expectations. ... ok
6: test_processes_exit_codes (rcutils.TestLoggingOutputFormatAfterShutdown)
6: Test that all executables finished cleanly. ... ok
6: 
6: ----------------------------------------------------------------------
6: Ran 2 tests in 0.003s
6: 
6: OK
```

clalancette, 2020-02-14:
This failure is a bit weird, since I don't see it in the CI nightlies or locally.  That leads me to suspect that it may be cyclonedds-specific; I'm going to kick off a local build to see if I can reproduce it that way.

rotu, 2020-02-14:
@clalancette I see these failures on the Linux nightly, even on the last stable build, and even with CycloneDDS disabled. Do a search for "Task exception was never retrieved":

https://ci.ros2.org/view/nightly/job/nightly_linux_release/lastStableBuild/consoleText

clalancette, 2020-02-14:
> @clalancette I see these failures on the Linux nightly, even on the last stable build, and even with CycloneDDS disabled --- it does not report as a test failure. Do a search for "Task exception was never retrieved":

Oops, yeah, my eyes slipped right over it since it wasn't a failure.  Investigating...

clalancette, 2020-02-14:
All right.  This turns out to be a bug in launch, though there is a way to workaround it in the test itself.

The bug basically comes down to the fact that this particular test doesn't have any tests that run concurrently with the processes under test.  It only has a test that runs after shutdown.  In that scenario, launch_testing seemingly fires up some kind of "dummy" unittest case, which completes immediately because it has no tests.  The rest of launch_testing then gets confused because a process under its control fired up and quit, but it has no record of it.  I'm going to file a bug against launch_testing for that.

In the meantime, we can workaround this particular failure by adding a test that runs in parallel with the processes, just waiting for them to complete.  I'll open a PR for that presently as well.

clalancette, 2020-02-14:
https://github.com/ros2/launch/issues/380 is the new bug under launch.

#209 is a workaround for this until the launch bug gets fixed.

rotu, 2020-02-15:
@clalancette thanks for demystifying this!

PR #209 (2020-02-18): Add in a concurrent test to test_logging_output_format.py
Before this PR, there was only a post_shutdown_test.  Due
to a bug in launch_testing, having only a post_shutdown test
causes stack traces to be printed while starting up the tests.
Add in this workaround to quiet the problem until the bug
is fixed.

Fixes #204
closed 2020-02-18 by clalancette