ros2/rcutils #220: RCUTILS_SET_ERROR_MSG use

adamsj-oci, 2020-04-01:
According to doxygen and this [line](https://github.com/ros2/rcutils/blob/master/include/rcutils/error_handling.h#L194), `RCUTILS_SET_ERROR_MSG()` should produce output if `rcutils_reset_error` wasn't called and this library was built with the `RCUTILS_REPORT_ERROR_HANDLING_ERRORS` flag set. I see that the flag is set [here](https://github.com/ros2/rcutils/blob/master/src/allocator.c#L28) and [here](https://github.com/ros2/rcutils/blob/master/src/error_handling_helpers.h#L25).  Since there are many of the `RCUTILS_SET_ERROR_MSG` (and `RMW_SET_ERROR_MSG`) calls in RCL and RMW, I wouldn't expect to see the stderr prints occur at all in run time unless turned by some means. What am I missing to turn these prints on?

hidmic, 2020-04-08:
@adamsj-oci the feature you describe ensures that `RCUTILS_SET_ERROR_MSG()` does not silently override previously set errors not handled explicitly prior to reset. It also ensures some invalid arguments in low-level APIs do not go by unnoticed, and thus it's unlikely you'll ever see such prints.

So I'm not entirely sure what you're trying to achieve here, but I'll go ahead and guess you want to get those error messages printed to `stderr` for debugging purposes. There's no way to do that. But considering that both `rcl` and `rmw`, as well as client libraries above in the stack, rigorously handle all errors, if you ever hit one of those you'll know. AFAIK the few places that "eat up" errors (e.g. because there's no caller to return it to) do print it before doing so.

adamsj-oci, 2020-04-08:
@hidmic Thanks for the feedback. 

I don't really mind if they go to `stderr`. It's obvious that the string provided to `RCUTILS_SET_ERROR_MSG` will output somewhere under some condition. [This line](https://github.com/ros2/rmw_fastrtps/blob/master/rmw_fastrtps_shared_cpp/src/publisher.cpp#L39) has no other purpose other than to output text. Where does this output appear? When is it active such that it will appear?

hidmic, 2020-04-08:
TL; DR it is up to calling code to retrieve it and do something about it. Or not.

---
 
`rcutils` error handling is somewhat similar to POSIX `errno` based error handling. There's a global error state, that in this case is populated with an arbitrary string, and functions let calling code know there was an error in some way -- often using a well-known numeric return code, but sometimes it may be something else. From then on, it is up to calling code to decide on what should be done with that information. In `rclpy` and `rclcpp` APIs implementation that use `rcl`, errors like these are usually turned into exceptions (so that eventually that string you mention becomes the message of an `std::runtime_error` exception).

hidmic, 2020-04-08:
@adamsj-oci for this sort of questions, [answers.ros.org](https://answers.ros.org/) is a better place. There, this discussion remains in the open for everyone to see. Here, it adds to the already humongous ROS 2 backlog and it will eventually be forgotten once this issue is closed. 

So I will close this issue and encourage you to continue the discussion over there. Hope you understand.

closed 2020-04-08 by hidmic

adamsj-oci, 2020-04-17:
As [requested](https://answers.ros.org/question/348867/rcutils_set_error_msg-and-rmw_set_error_msg-output-and-behavior/) , but this issue will likely never be answered there. It's part of this repo. It would be nice to document where and when to expect the text output more clearly in the doxygen output (error_handling.h) without having to dig the code to figure out how it works.

hidmic, 2020-04-24:
I've replied. We can better document `rcutils` error handling for sure -- and a contribution of yours in this regard would be more than welcomed. 

But I sense you still expect to be able to flip some switch and get all `*_SET_ERROR_MSG` messages dumped somewhere, and that's not how `rcutils` error handling works.

adamsj-oci, 2020-04-28:
Thanks for the update. Since this is not how `rcutils` works, I'm wondering if it makes sense to have a separate tool extract the error state even if the calling code does not?