ros2/rcutils #254: Memory leak in shared_library.c rcutils_load_shared_library

brawner, 2020-05-21:
It's possible there are other leaks in rcutils, but I wanted to ticket this one while I had the file open.

The file path created by rcutils_strdup in this function is not being deallocated in the RCUTILS_RET_OK path.

https://github.com/ros2/rcutils/blob/a67ad26cc60b0af46316d4ac96aae7d934ce9991/src/shared_library.c#L59

clalancette, 2020-05-22:
Yikes, that's a bad one.  We should probably fix that for Foxy.

brawner, 2020-05-22:
Actually now that I look at this closer, it's up to the calling code to deallocate the string path. Trying to deallocate this in `rcutils_load_shared_library` results in a double free. I think the code I was working in needed to be fixed. I'll keep hunting this one, but I think rcutils is perfectly fine.

closed 2020-05-22 by brawner