ros2/rcutils #310: MSBuild warnings C5105: macro expansion producing 'defined' has undefined behavior

jacobperron, 2020-11-11:
Building the latest versions of Eloquent or Foxy from source produces several warnings on Windows with the following message:

    C5105: macro expansion producing 'defined' has undefined behavior

For example (foxy): [image]

Seems to be coming from the following rcutils source files on the `foxy` branch:

- rcutils\src\filesystem.c
- rcutils\src\logging.c
- rcutils\src\process.c
- rcutils\src\shared_library.c
- rcutils\src\testing\fault_injection.c
- rcutils\src\time_win32.c

I have not observed the warning on Dashing or Rolling.

clalancette, 2020-12-03:
Ah yeah, we probably need to backport #311 to fix that.

clalancette, 2020-12-10:
I backported this to Foxy.  I'm not going to bother with Eloquent since this is the last sync, so I'll close this out.

PR #319 (2020-12-03): [Foxy backport]: Disable a Windows platform warning. (#311)

This closes #310 .
closed 2020-12-10 by clalancette