ros2/rcutils #329: Compilation error with rcutils in ros2

harshilbhatt2001, 2021-02-12:
**Required Info:**

- Operating System:
Arch Linux
- Installation type:
from Source
- Version or commit hash:
ROS 2 foxy
- DDS implementation:
Fast-RTPS (default)
- gcc 
Version 10.2

I get the following error while trying to build ros
```
 --- stderr: rcutils
CMake Deprecation Warning at /opt/ros2/foxy/src/gtest_vendor/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at /opt/ros2/foxy/src/gmock_vendor/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

In file included from /home/harshil/.local/src/ros2/src/ros2/rcutils/test/./mocking_utils/filesystem.hpp:35,
                 from /home/harshil/.local/src/ros2/src/ros2/rcutils/test/test_filesystem.cpp:22:
/home/harshil/.local/src/ros2/src/ros2/rcutils/test/./mocking_utils/filesystem.hpp:158:32: error: ‘__xstat’ was not declared in this scope; did you mean ‘_xstate’?
  158 |   MOCKING_UTILS_PATCH_TYPE(ID, __xstat) __xstat_mock_;
      |                                ^~~~~~~
/home/harshil/.local/src/ros2/src/ros2/rcutils/test/./mocking_utils/patch.hpp:316:51: note: in definition of macro ‘MOCKING_UTILS_PATCH_TYPE’
  316 |   decltype(mocking_utils::make_patch<id, decltype(function)>("", nullptr))
      |                                                   ^~~~~~~~
/home/harshil/.local/src/ros2/src/ros2/rcutils/test/./mocking_utils/filesystem.hpp:158:32: error: ‘__xstat’ was not declared in this scope; did you mean ‘_xstate’?
  158 |   MOCKING_UTILS_PATCH_TYPE(ID, __xstat) __xstat_mock_;
      |                                ^~~~~~~
/home/harshil/.local/src/ros2/src/ros2/rcutils/test/./mocking_utils/patch.hpp:316:51: note: in definition of macro ‘MOCKING_UTILS_PATCH_TYPE’
  316 |   decltype(mocking_utils::make_patch<id, decltype(function)>("", nullptr))
      |                                                   ^~~~~~~~
In file included from /home/harshil/.local/src/ros2/src/ros2/rcutils/test/test_filesystem.cpp:22:
/home/harshil/.local/src/ros2/src/ros2/rcutils/test/./mocking_utils/filesystem.hpp: In instantiation of ‘mocking_utils::filesystem::FileSystem<ID>::FileSystem(const string&) [with long unsigned int ID = 5; std::string = std::__cxx11::basic_string<char>]’:
/home/harshil/.local/src/ros2/src/ros2/rcutils/test/test_filesystem.cpp:216:30:   required from here
/home/harshil/.local/src/ros2/src/ros2/rcutils/test/./mocking_utils/filesystem.hpp:101:41: error: using invalid field ‘mocking_utils::filesystem::FileSystem<ID>::__xstat_mock_’
  101 |       MOCKING_UTILS_PATCH_PROXY(__xstat))
      |                                         ^
/home/harshil/.local/src/ros2/src/ros2/rcutils/test/./mocking_utils/filesystem.hpp:103:5: error: using invalid field ‘mocking_utils::filesystem::FileSystem<ID>::__xstat_mock_’
  103 |     __xstat_mock_.then_call(
      |     ^~~~~~~~~~~~~
/home/harshil/.local/src/ros2/src/ros2/rcutils/test/./mocking_utils/filesystem.hpp: In instantiation of ‘mocking_utils::filesystem::FileSystem<ID>::FileSystem(const string&) [with long unsigned int ID = 6; std::string = std::__cxx11::basic_string<char>]’:
/home/harshil/.local/src/ros2/src/ros2/rcutils/test/test_filesystem.cpp:253:30:   required from here
/home/harshil/.local/src/ros2/src/ros2/rcutils/test/./mocking_utils/filesystem.hpp:101:41: error: using invalid field ‘mocking_utils::filesystem::FileSystem<ID>::__xstat_mock_’
  101 |       MOCKING_UTILS_PATCH_PROXY(__xstat))
      |                                         ^
/home/harshil/.local/src/ros2/src/ros2/rcutils/test/./mocking_utils/filesystem.hpp:103:5: error: using invalid field ‘mocking_utils::filesystem::FileSystem<ID>::__xstat_mock_’
  103 |     __xstat_mock_.then_call(
      |     ^~~~~~~~~~~~~
/home/harshil/.local/src/ros2/src/ros2/rcutils/test/./mocking_utils/filesystem.hpp: In instantiation of ‘mocking_utils::filesystem::FileSystem<ID>::FileSystem(const string&) [with long unsigned int ID = 7; std::string = std::__cxx11::basic_string<char>]’:
/home/harshil/.local/src/ros2/src/ros2/rcutils/test/test_filesystem.cpp:281:30:   required from here
/home/harshil/.local/src/ros2/src/ros2/rcutils/test/./mocking_utils/filesystem.hpp:101:41: error: using invalid field ‘mocking_utils::filesystem::FileSystem<ID>::__xstat_mock_’
  101 |       MOCKING_UTILS_PATCH_PROXY(__xstat))
      |                                         ^
/home/harshil/.local/src/ros2/src/ros2/rcutils/test/./mocking_utils/filesystem.hpp:103:5: error: using invalid field ‘mocking_utils::filesystem::FileSystem<ID>::__xstat_mock_’
  103 |     __xstat_mock_.then_call(
      |     ^~~~~~~~~~~~~
/home/harshil/.local/src/ros2/src/ros2/rcutils/test/./mocking_utils/filesystem.hpp: In instantiation of ‘mocking_utils::filesystem::FileSystem<ID>::FileSystem(const string&) [with long unsigned int ID = 8; std::string = std::__cxx11::basic_string<char>]’:
/home/harshil/.local/src/ros2/src/ros2/rcutils/test/test_filesystem.cpp:377:30:   required from here
/home/harshil/.local/src/ros2/src/ros2/rcutils/test/./mocking_utils/filesystem.hpp:101:41: error: using invalid field ‘mocking_utils::filesystem::FileSystem<ID>::__xstat_mock_’
  101 |       MOCKING_UTILS_PATCH_PROXY(__xstat))
      |                                         ^
/home/harshil/.local/src/ros2/src/ros2/rcutils/test/./mocking_utils/filesystem.hpp:103:5: error: using invalid field ‘mocking_utils::filesystem::FileSystem<ID>::__xstat_mock_’
  103 |     __xstat_mock_.then_call(
      |     ^~~~~~~~~~~~~
In file included from /usr/include/c++/10.2.0/functional:59,
                 from /opt/ros2/foxy/src/gtest_vendor/include/gtest/gtest-printers.h:103,
                 from /opt/ros2/foxy/src/gtest_vendor/include/gtest/gtest-matchers.h:46,
                 from /opt/ros2/foxy/src/gtest_vendor/include/gtest/internal/gtest-death-test-internal.h:39,
                 from /opt/ros2/foxy/src/gtest_vendor/include/gtest/gtest-death-test.h:41,
                 from /opt/ros2/foxy/src/gtest_vendor/include/gtest/gtest.h:62,
                 from /home/harshil/.local/src/ros2/src/ros2/rcutils/test/test_filesystem.cpp:15:
/usr/include/c++/10.2.0/bits/std_function.h:601:7: error: ‘std::function<_Res(_ArgTypes ...)>::function(_Functor) [with _Functor = mocking_utils::filesystem::FileSystem<ID>::FileSystem<5>::<lambda(auto:1&& ...)>; <template-parameter-2-2> = void; <template-parameter-2-3> = void; _Res = __dirstream*; _ArgTypes = {const char*}]’, declared using local type ‘mocking_utils::filesystem::FileSystem<ID>::FileSystem<5>::<lambda(auto:1&& ...)>’, is used but never defined [-fpermissive]
  601 |       function<_Res(_ArgTypes...)>::
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/10.2.0/bits/std_function.h:601:7: error: ‘std::function<_Res(_ArgTypes ...)>::function(_Functor) [with _Functor = mocking_utils::filesystem::FileSystem<ID>::FileSystem<6>::<lambda(auto:1&& ...)>; <template-parameter-2-2> = void; <template-parameter-2-3> = void; _Res = __dirstream*; _ArgTypes = {const char*}]’, declared using local type ‘mocking_utils::filesystem::FileSystem<ID>::FileSystem<6>::<lambda(auto:1&& ...)>’, is used but never defined [-fpermissive]
/usr/include/c++/10.2.0/bits/std_function.h:601:7: error: ‘std::function<_Res(_ArgTypes ...)>::function(_Functor) [with _Functor = mocking_utils::filesystem::FileSystem<ID>::FileSystem<7>::<lambda(auto:1&& ...)>; <template-parameter-2-2> = void; <template-parameter-2-3> = void; _Res = __dirstream*; _ArgTypes = {const char*}]’, declared using local type ‘mocking_utils::filesystem::FileSystem<ID>::FileSystem<7>::<lambda(auto:1&& ...)>’, is used but never defined [-fpermissive]
/usr/include/c++/10.2.0/bits/std_function.h:601:7: error: ‘std::function<_Res(_ArgTypes ...)>::function(_Functor) [with _Functor = mocking_utils::filesystem::FileSystem<ID>::FileSystem<8>::<lambda(auto:1&& ...)>; <template-parameter-2-2> = void; <template-parameter-2-3> = void; _Res = __dirstream*; _ArgTypes = {const char*}]’, declared using local type ‘mocking_utils::filesystem::FileSystem<ID>::FileSystem<8>::<lambda(auto:1&& ...)>’, is used but never defined [-fpermissive]
make[2]: *** [CMakeFiles/test_filesystem.dir/build.make:82: CMakeFiles/test_filesystem.dir/test/test_filesystem.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1162: CMakeFiles/test_filesystem.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:160: all] Error 2
 ---
Failed   <<< rcutils [30.5s, exited with code 2]

```

clalancette, 2021-02-12:
@harshilbhatt2001 Arch Linux isn't one of our target platforms, so it isn't totally surprising that it doesn't work there.  That being said, if you find a solution to make it compile we are more than happy to review a PR.

Kurtoid, 2021-02-16:
I was only able to build it with `colcon build --cmake-args " -DBUILD_TESTING=0"`

paeunt, 2021-03-02:
Fixed in https://github.com/ros2/rcutils/pull/330, cherrypick this change and have a try please.

clalancette, 2022-04-22:
Given that this seemed solved a while ago, I'm going to close this out.  Feel free to reopen if that is not the case.

closed 2022-04-22 by clalancette