ros2/rcutils #348: Warnings while compiling to WebAssembly

allsey87, 2021-12-27:
Hi! I am working on compiling ROS2 nodes to WebAssembly using Emscripten so that I can simulate robot control software in the browser. For the most part, this is going smoothly so far but there are a couple warnings that I think should be addressed:

1. Conditional compilation in [`rcutils_get_platform_library_name`](https://github.com/ros2/rcutils/blob/67095577eaf39977f30f7db1ae92f7b0b2ec48d0/src/shared_library.c#L286)
> rcutils/src/shared_library.c:289:16: warning: unused parameter 'buffer_size' [-Wunused-parameter]
> rcutils/src/shared_library.c:290:8: warning: unused parameter 'debug' [-Wunused-parameter]

It seems when I compile using Clang/Emscripten, none of the `#ifdef`s are matching my configuration in `rcutils_get_platform_library_name`, which is a bit unusual since I was under the impression that Emscripten defines `__LINUX__` (uppercase). I will investigate this, however, there is also the possibility of checking whether `__EMSCRIPTEN__` has been defined.

2. Conditional compilation in [`rcutils_strerror`](https://github.com/ros2/rcutils/blob/67095577eaf39977f30f7db1ae92f7b0b2ec48d0/src/strerror.c#L26)
> rcutils/src/strerror.c:32:10: warning: incompatible integer to pointer conversion initializing 'char *' with an expression of type 'int' [-Wint-conversion]

This warning seems very similar to the previous one. It seems again that conditional compilation has gone down the wrong path since the version `strerror_r` that comes with Emscripten's C standard library returns an `int` instead of a `char*`.

For reference, I am using the Emscripten 3.1.0 (Clang 14.0.0) with the following colcon mixin:
```json
{
    "build": {
        "emscripten": {
            "cmake-args": [
                "-DCMAKE_TOOLCHAIN_FILE=/path/to/emsdk/upstream/emscripten/cmake/Modules/Platform/Emscripten.cmake",
                "-DCMAKE_STAGING_PREFIX=/path/to/developer/emsdk/upstream/emscripten/cache/sysroot/usr/",
                "-DBUILD_SHARED_LIBS=OFF",
                "-DBUILD_TESTING=OFF",
                "--no-warn-unused-cli"
            ]
        }
    }
}
```

allsey87, 2021-12-27:
Note: Emscripten does support [dynamic linking](https://emscripten.org/docs/compiling/Dynamic-Linking.html) (with `so` or `wasm` as the recommended extension), however, it is still a bit strange that [`shared_library.c`](https://github.com/ros2/rcutils/blob/67095577eaf39977f30f7db1ae92f7b0b2ec48d0/src/shared_library.c) is getting compiled at all when `BUILD_SHARED_LIBS=OFF`...

aprotyas, 2021-12-27:
Just for reference - since WASM is a separate virtual ISA, it might be helpful to look at what was done for other platforms (like QNX): https://github.com/ros2/ros2/issues/988.

1. For point 1, as you alluded, another condition can be added to
https://github.com/ros2/rcutils/blob/67095577eaf39977f30f7db1ae92f7b0b2ec48d0/src/shared_library.c#L297

2. For point 2, `rcutils_strerror` does have a conditional branch that deals with the [XSI-compliant `strerror_r` signature](https://linux.die.net/man/3/strerror_r) (second signature), but I think conditional compilation went down the wrong path here too.

https://github.com/ros2/rcutils/blob/67095577eaf39977f30f7db1ae92f7b0b2ec48d0/src/strerror.c#L37-L44