ros2/rcutils #358: Why does `rcutils_expand_user` fail on non-existent directory?

EricCousineau-TRI, 2022-05-10:
For downstream context, see: https://github.com/robotlocomotion/drake-ros/pull/79#pullrequestreview-965487607
Basically, we had a read-only filesystem, and failure occurred when trying to expand a path.

If this trace is correct, then (IMO) this seems to violate Single Respnosibility Principle - expanding paths should *only* care about expand paths, not whether the path exists or not.

\cc @sloretz

clalancette, 2022-05-10:
Just taking a look at the code for `rcutils_expand_user`, it does not seem to care about whether the directory exists: https://github.com/ros2/rcutils/blob/a6fa70a7d53da20c569c1c0187ac379405f02047/src/filesystem.c#L207-L228 .  That said, it *does* care that the HOME environment variable is set (via the call to `rcutils_get_home_dir`.  Could that be the issue you are seeing?

EricCousineau-TRI, 2022-05-11:
Ahh, I think that was it! Confirmed locally in https://github.com/RobotLocomotion/drake-ros/compare/c7c235a9...2427b5af4dd

In this case, I think it was `ROS_HOME` not being set that caused the `rclutils_expand_user` traceback.

Sorry for the noise, and thanks for clarifying!

closed 2022-05-11 by EricCousineau-TRI