ros2/rcutils #364: High logging overhead

almaslov, 2022-05-24:
Hello!
I was profiling some nodes in my project recently and noticed an issue in case of high incoming messages rate(~2-3K hz). In this case `rcl_wait` consumes significant amount of CPU. About 30-50% of `rcl_wait` execution time is `rcutils_logging_logger_is_enabled_for`:
[image]
It looks like most of overhead can be eliminated removing `rcutils_logging_severity_level_from_string` and switching to integral `log_level` as mentioned in comment [here](https://github.com/ros2/rcutils/blob/master/src/logging.c#L410) and [here](https://github.com/ros2/rcutils/blob/master/src/logging.c#L481)

almaslov, 2022-05-24:
I would be glad to make a PR, but I'm a little confused with idea of implementing string-to-int map type which is almost complete copy-paste of string-to-string map. Could you suggest, which solution did you have in mind?

clalancette, 2022-05-25:
I took a quick look at this as well.  I think there are several problems in here:

1.  `rcutils_logging_severity_level_from_string` is doing a string copy and toupper on the string every time it is invoked (https://github.com/ros2/rcutils/blob/8da2aace17bcecad3bdf0bebafe23298602f50c0/src/logging.c#L320-L355).  While I guess that is nice, I would argue that we can just make this case-sensitive and save a lot of work here.  Since this is changing a semantic, we'd want to some research on how widely used it is, but I think that should be straightforward.
2. As @fujitatomoya pointed out, we are doing a lot of work in `rcutils_logging_get_logger_effective_level` to try to suss out the parts of the logger that we want.  We may be able to optimize this by memoizing that result, though we would still incur overhead in checking on the memoization.  We could also reconsider whether we really need that functionality.  That is more controversial, since I can definitely see uses for the current system, but logging really should be as simple as possible.

I'm not sure I understand your latest proposal.  If we had some kind of global logging update timestamp, wouldn't we just be throttling messages?  While that certainly would help, we already have the `THROTTLE` versions of the messages.

almaslov, 2022-05-26:
@clalancette Thanks for reply.
I agree, that `rcutils_logging_severity_level_from_string` has large overhead by copying string and it could be fixed with case-insensitive comparison. But my initial thought was that logging macros shouldn't use (in nested invocations) `rcutils_logging_severity_level_from_string` at all. 

`logging.h` API uses integer `log_level` in all functions(except `rcutils_logging_severity_level_from_string`). As far as I can see, integer `log_level` is converted to string value only for storage purpose. And this is a detail of implementation, not public API. So storing integer value instead of string value could save CPU from unnecessary conversions.

>I'm not sure I understand your latest proposal. If we had some kind of global logging update timestamp, wouldn't we just be throttling messages? While that certainly would help, we already have the THROTTLE versions of the messages.

This global logging update timestamp, say `last_global_logging_update`, should be updated on logging configuration change, e.g. when `rcutils_logging_set_default_logger_level` or `rcutils_logging_set_logger_level` called. If logging configuration changes, `rcutils_logging_logger_is_enabled_for` can be recalculated only one time, until next configuration update. The idea is `last_global_logging_update` affects only `rcutils_logging_logger_is_enabled_for` calls, not logging handler calls. So, as you can see in my code snippet I added condition that checks if `last_global_logging_update` changed since last macro call.

Actually now I see that there is no way to handle global log level change trough `g_rcutils_logging_default_logger_level` global variable which is exposed in `logging.h`. So I'm not sure this proposal can be implemented correctly.

closed 2022-09-07 by clalancette