ros2/rcutils #365: AddressSanitizer: new-delete-type-mismatch in rcutils_string_map_fini

tylerjw, 2022-06-06:
Here is the output from Asan I'm now getting in my tests. I did not have these errors in Galactic but am now experiencing them after upgrading from Galactic on Focal to Rolling on Jammy.

```
build/core/Testing/20220606-1842/Test.xml: 3 tests, 0 errors, 1 failure, 0 skipped
- test_parameters
  <<< failure message
    -- run_test.py: invoking following command in '/home/runner/work/project/project/.work/target_ws/build/core':
     - /home/runner/work/project/project/.work/target_ws/build/core/test_parameters -r junit -o /home/runner/work/project/project/.work/target_ws/build/core/test_results/core/test_parameters.catch2.xml
    =================================================================
    ==8184==ERROR: AddressSanitizer: new-delete-type-mismatch on 0x607000011f20 in thread T0:
      object passed to delete has wrong type:
      size of the allocated type:   72 bytes;
      size of the deallocated type: 1 bytes.
        #0 0x7fee6b27a22f in operator delete(void*, unsigned long) ../../../../src/libsanitizer/asan/asan_new_delete.cpp:172
        #1 0x7fee6ac52dae in rcutils_string_map_fini (/opt/ros/rolling/lib/librcutils.so+0x9dae)
        #2 0x7fee6ae753a5  (/opt/ros/rolling/lib/librcl.so+0x233a5)
        #3 0x7fee6ae756c7 in rcl_node_resolve_name (/opt/ros/rolling/lib/librcl.so+0x236c7)
        #4 0x7fee6ae75898 in rcl_publisher_init (/opt/ros/rolling/lib/librcl.so+0x23898)
        #5 0x7fee6afd932d in rclcpp::PublisherBase::PublisherBase(rclcpp::node_interfaces::NodeBaseInterface*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rosidl_message_type_support_t const&, rcl_publisher_options_s const&) (/opt/ros/rolling/lib/librclcpp.so+0x14232d)
        #6 0x7fee6afac7fb  (/opt/ros/rolling/lib/librclcpp.so+0x1157fb)
        #7 0x7fee6afacf30 in std::_Function_handler<std::shared_ptr<rclcpp::PublisherBase> (rclcpp::node_interfaces::NodeBaseInterface*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::QoS const&), rclcpp::create_publisher_factory<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void>, rclcpp::Publisher<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void> > >(rclcpp::PublisherOptionsWithAllocator<std::allocator<void> > const&)::{lambda(rclcpp::node_interfaces::NodeBaseInterface*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::QoS const&)#1}>::_M_invoke(std::_Any_data const&, rclcpp::node_interfaces::NodeBaseInterface*&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::QoS const&) (/opt/ros/rolling/lib/librclcpp.so+0x115f30)
        #8 0x7fee6af9a2ce in rclcpp::node_interfaces::NodeParameters::NodeParameters(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeLoggingInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeTopicsInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeServicesInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeClockInterface>, std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter> > const&, bool, bool, rclcpp::QoS const&, rclcpp::PublisherOptionsBase const&, bool, bool) (/opt/ros/rolling/lib/librclcpp.so+0x1032ce)
        #9 0x7fee6af92c95 in rclcpp::Node::Node(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::NodeOptions const&) (/opt/ros/rolling/lib/librclcpp.so+0xfbc95)
        #10 0x7fee6af934c7 in rclcpp::Node::Node(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::NodeOptions const&) (/opt/ros/rolling/lib/librclcpp.so+0xfc4c7)
        #11 0x555af062e6c9 in decltype (::new ((void*)(0)) rclcpp::Node((declval<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&>)())) std::construct_at<rclcpp::Node, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&>(rclcpp::Node*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) /usr/include/c++/11/bits/stl_construct.h:97
        #12 0x555af062e9dc in void std::allocator_traits<std::allocator<rclcpp::Node> >::construct<rclcpp::Node, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&>(std::allocator<rclcpp::Node>&, rclcpp::Node*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) /usr/include/c++/11/bits/alloc_traits.h:518
        #13 0x555af062dd8b in std::_Sp_counted_ptr_inplace<rclcpp::Node, std::allocator<rclcpp::Node>, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&>(std::allocator<rclcpp::Node>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) /usr/include/c++/11/bits/shared_ptr_base.h:519
        #14 0x555af062d12b in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<rclcpp::Node, std::allocator<rclcpp::Node>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&>(rclcpp::Node*&, std::_Sp_alloc_shared_tag<std::allocator<rclcpp::Node> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) /usr/include/c++/11/bits/shared_ptr_base.h:650
        #15 0x555af062c9b1 in std::__shared_ptr<rclcpp::Node, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<rclcpp::Node>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&>(std::_Sp_alloc_shared_tag<std::allocator<rclcpp::Node> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) /usr/include/c++/11/bits/shared_ptr_base.h:1342
        #16 0x555af062ab37 in std::shared_ptr<rclcpp::Node>::shared_ptr<std::allocator<rclcpp::Node>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&>(std::_Sp_alloc_shared_tag<std::allocator<rclcpp::Node> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) /usr/include/c++/11/bits/shared_ptr.h:409
        #17 0x555af06274d2 in std::shared_ptr<rclcpp::Node> std::allocate_shared<rclcpp::Node, std::allocator<rclcpp::Node>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&>(std::allocator<rclcpp::Node> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) /usr/include/c++/11/bits/shared_ptr.h:863
        #18 0x555af06233f7 in std::shared_ptr<rclcpp::Node> std::make_shared<rclcpp::Node, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) /usr/include/c++/11/bits/shared_ptr.h:879
        #19 0x555af0614060 in make_node /home/runner/work/project/project/.work/target_ws/src/project/core/test/test_parameter_loader.cpp:27
        #20 0x555af061507a in C_A_T_C_H_T_E_S_T_0 /home/runner/work/project/project/.work/target_ws/src/project/core/test/test_parameter_loader.cpp:34
        #21 0x555af028e04f in Catch::TestInvokerAsFunction::invoke() const /opt/upstream/install/catch2/include/catch2/catch.hpp:14328
        #22 0x555af028833f in Catch::TestCase::invoke() const /opt/upstream/install/catch2/include/catch2/catch.hpp:14167
        #23 0x555af02651a6 in Catch::RunContext::invokeActiveTestCase() /opt/upstream/install/catch2/include/catch2/catch.hpp:13027
        #24 0x555af02635ca in Catch::RunContext::runCurrentTest(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) /opt/upstream/install/catch2/include/catch2/catch.hpp:12992
        #25 0x555af0257f4a in Catch::RunContext::runTest(Catch::TestCase const&) /opt/upstream/install/catch2/include/catch2/catch.hpp:12761
        #26 0x555af02703d7 in execute /opt/upstream/install/catch2/include/catch2/catch.hpp:13354
        #27 0x555af0278ba8 in Catch::Session::runInternal() /opt/upstream/install/catch2/include/catch2/catch.hpp:13560
        #28 0x555af0277151 in Catch::Session::run() /opt/upstream/install/catch2/include/catch2/catch.hpp:13516
[... cut ...]
        #4 0x7f1204244ff9 in void* rclcpp::allocator::retyped_allocate<std::allocator<char> >(unsigned long, void*) /opt/ros/rolling/include/rclcpp/rclcpp/allocator/allocator_common.hpp:39
        #5 0x7f1202d1bbb2 in rcutils_string_map_init (/opt/ros/rolling/lib/librcutils.so+0x9bb2)
    SUMMARY: AddressSanitizer: new-delete-type-mismatch ../../../../src/libsanitizer/asan/asan_new_delete.cpp:172 in operator delete(void*, unsigned long)
    ==8306==HINT: if you don't care about these errors you may set ASAN_OPTIONS=new_delete_type_mismatch=0
    ==8306==ABORTING
    -- run_test.py: return code 1
    -- run_test.py: verify result file 
```

fujitatomoya, 2022-06-06:
probably related to https://github.com/ros2/rclcpp/pull/1324

tylerjw, 2022-06-06:
I created a minimal project here that demonstrates this problem:
https://github.com/tylerjw/ros2_sanitize

Here is a CI run that shows it is broken in main and testing of both Humble and Rolling but in Galactic you can create a rclcpp::Node without asan errors.
https://github.com/tylerjw/ros2_sanitize/runs/6763851100?check_suite_focus=true

tylerjw, 2022-06-06:
Closing to re-open in rclcpp as in my example I'm creating a rclcpp::Node and it is related to a bug in rclcpp.

closed 2022-06-06 by tylerjw