ros2/rcutils #373: "Failed to find exported target names" error while colcon build

Aki1608, 2022-08-12:
Hi I am trying to build this package inside docker. I am using [this](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/blob/release-dp/docker/Dockerfile.humble) docker image. The docker is using Ubuntu 20.04 as base OS. While executing colcon build command, it is giving following error. What can be the reason behind this?

`CMake Error at /opt/ros/humble/install/share/performance_test_fixture/cmake/ament_cmake_export_targets-extras.cmake:18 (message):
  Failed to find exported target names in
  '/opt/ros/humble/install/share/performance_test_fixture/cmake/performance_test_fixtureExport.cmake'
Call Stack (most recent call first):
  /opt/ros/humble/install/share/performance_test_fixture/cmake/performance_test_fixtureConfig.cmake:41 (include)
  CMakeLists.txt:150 (find_package)`

gavanderhoorn, 2022-08-15:
Is this https://github.com/ament/ament_cmake/issues/396?

clalancette, 2022-08-15:
> Is this [ament/ament_cmake#396](https://github.com/ament/ament_cmake/issues/396)?

Yes, good call.  That is almost certainly the case.  I'm going to close this out since we merged in that PR, but feel free to reopen if you are still having this problem.

closed 2022-08-15 by clalancette

Aki1608, 2022-08-15:
Yes. That was the error. Thank you so much. :) @gavanderhoorn