ros2/rcutils #388: Undefined symbol: rcutils_log_internal

tommy-erko, 2022-09-30:
Hi, following the ROS2 Rolling update yesterday (including update of ros-rolling-rcutils: 5.2.0-1 → 6.0.0-1), we started to have an issue with Gazebo not being able to find a logging utility from this repository. The error affects both old Gazebo "Classic" and new Gazebo (originally Ignition). We have an up-to-date version of both simulators corresponding to the new ros-rolling version.

Error logs:
Ignition Gazebo:
`[create-2] /opt/ros/rolling/lib/ros_ign_gazebo/create: symbol lookup error: /opt/ros/rolling/lib/ros_ign_gazebo/create: undefined symbol: rcutils_log_internal`
Gazebo Classic:
`[gzserver-1] gzserver: symbol lookup error: /opt/ros/rolling/lib/libgazebo_ros_imu_sensor.so: undefined symbol: rcutils_log_internal`

The result of the error is that none of the Gazebo versions is able to start.

Through a quick scan of the released commits in this repo, I suspects the error might have occured [here](https://github.com/ros2/rcutils/commit/f0bee9e66830ababfe1e0f2414d46e36a0b60ec5).

clalancette, 2022-09-30:
> Through a quick scan of the released commits in this repo, I suspects the error might have occured [here](https://github.com/ros2/rcutils/commit/f0bee9e66830ababfe1e0f2414d46e36a0b60ec5).

Yes, that is certainly where that symbol was introduced.

That said, everything in Rolling should have been rebuilt so it could find that symbol, so I'm not sure why that is occurring.  Here, if I just run `/opt/ros/rolling/lib/ros_ign_gazebo/create`, things seem to work for me.

Can you give me a self-contained example of the problem you are having?

tommy-erko, 2022-10-03:
Hi, it was proably a false alarm, we really weren't able to run our simulation in old Gazebo and Ignition Gazebo was falling inside Docker. But since saturday, it all runs perfectly. I suppose the new packages did not download/install properly at first. I will therefore close this, there is no error:)

closed 2022-10-03 by tommy-erko