ros2/rcutils #403: performance_test_fixture not found

mpconte, 2022-11-03:
On Ubuntu WSL2 subsystem on Windows 10, under the **humble** branch, when running:

```
mkdir build
cd build
cmake ..
```

I receive the following error:

```

-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found ament_cmake_python: 1.3.2 (/mnt/d/ros2-humble/install/share/ament_cmake_python/cmake)
-- Found ament_cmake_ros: 0.10.0 (/mnt/d/ros2-humble/install/share/ament_cmake_ros/cmake)
-- Found Python3: /usr/bin/python3.8 (found version "3.8.10") found components: Interpreter
-- Performing Test USES_GLIBC
-- Performing Test USES_GLIBC - Success
CMake Error at CMakeLists.txt:150 (find_package):
  By not providing "Findperformance_test_fixture.cmake" in CMAKE_MODULE_PATH
  this project has asked CMake to find a package configuration file provided
  by "performance_test_fixture", but CMake did not find one.

  Could not find a package configuration file provided by
  "performance_test_fixture" with any of the following names:

    performance_test_fixtureConfig.cmake
    performance_test_fixture-config.cmake

  Add the installation prefix of "performance_test_fixture" to
  CMAKE_PREFIX_PATH or set "performance_test_fixture_DIR" to a directory
  containing one of the above files.  If "performance_test_fixture" provides
  a separate development package or SDK, be sure it has been installed.

-- Configuring incomplete, errors occurred!
See also "/mnt/d/rcutils/build/CMakeFiles/CMakeOutput.log".
```

clalancette, 2022-11-03:
Have you built and installed `performance_test_fixture` first?

mpconte, 2022-11-03:
Hi @clalancette, yes I did install and it correctly updated the CMAKE_FIND_ROOT_PATH to include the path for it since then, but now I receive the following error:
```

CMake Error at /mnt/d/performance_test_fixture/build/ament_cmake_core/ament_cmake_export_targets-extras.cmake:9 (include):
  include could not find requested file:

    /mnt/d/performance_test_fixture/build/ament_cmake_core/performance_test_fixtureExport.cmake
Call Stack (most recent call first):
  /mnt/d/performance_test_fixture/build/ament_cmake_core/performance_test_fixtureConfig.cmake:41 (include)
  CMakeLists.txt:150 (find_package)

CMake Error at /mnt/d/performance_test_fixture/build/ament_cmake_core/ament_cmake_export_targets-extras.cmake:13 (file):
  file STRINGS file
  "/mnt/d/performance_test_fixture/build/ament_cmake_core/performance_test_fixtureExport.cmake"
  cannot be read.
Call Stack (most recent call first):
  /mnt/d/performance_test_fixture/build/ament_cmake_core/performance_test_fixtureConfig.cmake:41 (include)
  CMakeLists.txt:150 (find_package)

CMake Error at /mnt/d/performance_test_fixture/build/ament_cmake_core/ament_cmake_export_targets-extras.cmake:18 (message):
  Failed to find exported target names in
  '/mnt/d/performance_test_fixture/build/ament_cmake_core/performance_test_fixtureExport.cmake'
Call Stack (most recent call first):
  /mnt/d/performance_test_fixture/build/ament_cmake_core/performance_test_fixtureConfig.cmake:41 (include)
  CMakeLists.txt:150 (find_package)

```

clalancette, 2022-11-03:
So we don't test on WSL2, and you also aren't using our standard `colcon` tooling, so we probably can't help you.

That said, if you find out what the issue is we'd be happy to review patches to either make fixes or to make the documentation more clear on how to do this.

mpconte, 2022-11-03:
Sorry, I should have mentioned that in my previous post that I was using colcon to build exclusively rcutils and received this error (on Ubuntu WSL2)