ros2/rcutils #409: Superfluous `va_copy` call in `rcutils_char_array_vsprintf`

jrutgeer, 2023-02-17:
There seems to be a superfluous `va_copy` call in `rcutils_char_array_vsprintf`.

I cannot asses whether this has a relevant impact wrt logging perfomance, but given recent PR about optimization https://github.com/ros2/rcutils/pull/381 I assume it's worth mentioning to those who know better than me. :-)

If I understand correctly, `va_copy` copies `*args` into `args_clone` for `rcutils_char_array_vsprintf`:

https://github.com/ros2/rcutils/blob/17dbbec480d6155b65764b4ec4f773a1fc31e565/src/logging.c#L1313-L1321

However, `rcutils_char_array_vsprintf` doc states "The `va_list args` will be cloned before being used, so a user can safely use it again after calling this function."

This seems indeed correct, as `rcutils_char_array_vsprintf` passes `args` to `_rcutils_char_array_vsprintf`, which has its own `va_copy`:
 
https://github.com/ros2/rcutils/blob/17dbbec480d6155b65764b4ec4f773a1fc31e565/src/char_array.c#L166-L198

https://github.com/ros2/rcutils/blob/17dbbec480d6155b65764b4ec4f773a1fc31e565/src/char_array.c#L151-L164

So it might be a simple optimization to simply remove the `va_list / va_copy` in `rcutils_char_array_vsprintf`.

fujitatomoya, 2023-03-01:
@jrutgeer thanks for the issue and explanation. i believe that makes sense.

fujitatomoya, 2023-03-01:
both changes can be backported to humble.

closed 2023-03-02 by fujitatomoya