ros2/rcutils #425: Consol formating ignores escape sequences

SammyRamone, 2023-06-27:
The RCUTILS_CONSOLE_OUTPUT_FORMAT environment variable is a very nice way to format the log output of ROS 2 to one's own liking. However, it does not allow using any escape sequences (e.g. \n or any color codes) in it. It think it would be nice if this works, as this would, for example, allow to print different parts of the message in different colors or to structure the information over multiple lines.

Simple example of the issue:
```
➜ export RCUTILS_CONSOLE_OUTPUT_FORMAT='{name}\n {message}'
➜ echo $RCUTILS_CONSOLE_OUTPUT_FORMAT 
{name}
 {message}
➜ python3 test_log.py
test_node\n Hello World
```
The '\n' is not interpreted as a new line (ASCII code 10) but as an '\' and 'n' (ASCII code 92 and 110). This is the same for all special characters that are set in the environment variable. However, the echo command shows the newline correctly.

If environment variable is empty and the same format is set directly in the code as default format here
https://github.com/ros2/rcutils/blob/025f78fa7c03666b5fa1057af00a32ed8bbd64a0/src/logging.c#L87-L88
The behavior is correct and the newline is printed.
```
➜  ws_rcutils python3 test_log.py
test_node
 Hello World
```

It looks like this issue comes from getting the environment variable.
https://github.com/ros2/rcutils/blob/025f78fa7c03666b5fa1057af00a32ed8bbd64a0/src/env.c#L78
The return of this gives a char array which encodes the '\n' as '\' and 'n' characters.

I used zsh, but it also happens with bash. System is Ubuntu 22.04 with ROS rolling.

fujitatomoya, 2023-07-10:
@iuhilnehc-ynos can you take a look at this if you have bandwidth?

fujitatomoya, 2023-07-12:
@SammyRamone it would be appreciated if you can try https://github.com/ros2/rcutils/pull/426. thanks in advance.

SammyRamone, 2023-07-12:
> @SammyRamone it would be appreciated if you can try #426. thanks in advance.

I don't have time today, but I will try to do it tomorrow.

fujitatomoya, 2023-09-15:
@clalancette i will go ahead to close this issue since this is behavior change, i believe we do not backport this to iron or humble.

closed 2023-09-15 by fujitatomoya