ros2/rcutils #428: Setting/relying on _GNU_SOURCE seems incorrect

allsey87, 2023-08-10:
PR #267 broke compatibility with [musl libc](https://www.musl-libc.org/). I am seeing that `_GNU_SOURCE` is not enabled in `errno.h` which means that `program_invocation_name` is never defined.

As I understand, the programming standards for ROS2 should be C99 and C++17 (humble -> rolling). If this is the case, should this repo be relying on `_GNU_SOURCE` being defined in the first place?

clalancette, 2023-08-10:
> Moreover, regardless of whether we are targeting C99 or C11, it does not seem right to me that we are relying on `_GNU_SOURCE` being defined in either case.

I'm not super familiar with `program_invocation_name` here, but if there is another way to get access on it for both glibc and musl, we'd be happy to review a patch.

allsey87, 2023-08-10:
`program_invocation_name` is just used to implement `rcutils_get_executable_name` on some platforms. It comes from the header file errno.h, which under musl is only available if `_GNU_SOURCE` defined. The current test (check_c_compiler_uses_glibc.cmake), implemented in #267, doesn't seem to work for musl (I guess the backtrace function is not available?).

The problem is, defining `_GNU_SOURCE` for the whole rcutils package causes errors in other parts of the code, for example:

> ros2/rcutils/src/strerror.c:32:10: error: incompatible integer to pointer conversion initializing 'char *' with an expression of type 'int' [-Wint-conversion]
  char * msg = strerror_r(errno, buffer, buffer_length);

The only solution I have at the moment is to revert #267 (which also puts back a `#define _GNU_SOURCE` just before it `#include <error.h>`). When I see stuff like this, I cannot help but to think that ROS2, by trying to abstract over so many things, has failed a bit at the philosophy of "doing one thing well".

closed 2023-08-10 by allsey87