ros2/rcutils #472: Option for rcutils_set_env() to not overwrite existing environment variables.

Yadunund, 2024-07-16:
At present [rcutils_set_env](https://github.com/ros2/rcutils/blob/cf3e9986a557e4c893eb718b90c77ce189494a11/src/env.c#L50) always overwrites the value of the environment variable. However it would be nice to have the overwriting behavior configurable, ie, specify that `1` or `0` at the end of the `setenv` call.

This is useful in cases where the application would like to set the envar if previously unset in the environment.

Yadunund, 2024-07-17:
If the request is valid, should it be implemented as a new API or do we add a new `overwrite` argument thereby breaking API?

Barry-Xu-2018, 2024-07-17:
> If the request is valid, should it be implemented as a new API or do we add a new overwrite argument thereby breaking API?

For compatibility, I prefer using the new API and marking the old one as "DEPRECATED."

Like https://github.com/ros2/rcl/pull/1167/files

PR #473 (2024-09-16): Add new API to set envar while specifying overwrite
Fix #472
PR #480 (2024-09-17): Add new API to set envar while specifying overwrite (backport #473)
Fix #472 <hr>This is an automatic backport of pull request #473 done by [Mergify](https://mergify.com).
PR #481 (2024-09-17): Add new API to set envar while specifying overwrite (backport #473)
Fix #472 <hr>This is an automatic backport of pull request #473 done by [Mergify](https://mergify.com).