ros2/rcutils #475: logging: log4cxx support?

rr-tom-noble, 2024-09-02:
I'm currently using ROS noetic which I believe uses rosconsole / log4cxx for logging. This appears to be a widely used and customisable logging framework. As far as I can tell, it seems that ROS2 is using its own logging implementation? Is there a plan to support log4cxx in the future? If not, what was the reason for moving away from it?

rr-tom-noble, 2024-09-02:
Ah, I think I've found my answer here:
https://github.com/ros2/rcl_logging/pull/78

My use-case is that I'm wanting to output the logs in a JSON format. I've read the logging and config documentation here:
https://docs.ros.org/en/humble/Tutorials/Demos/Logging-and-logger-configuration.html

My use-case includes adding custom fields to the output, so I don't think I'd simply be able to make use of the `RCUTILS_CONSOLE_OUTPUT_FORMAT` environment variable.

From what I can tell, I may need to implement a custom logger, but it's unclear to me:

- Which class this should be inheriting from: `rclcpp::Logger` ? `spdlog::formatter` ?
- How I can configure ROS2 to use this custom logger by default (i.e. something akin to the `rosconsole` config file from ROS1)

rr-tom-noble, 2024-09-02:
https://github.com/gabime/spdlog/issues/2861

Something like this may work for my use-case. Is there a way to specify custom spdlog formatters in config?

fujitatomoya, 2024-09-03:
@rr-tom-noble i think that you already found the original query with this issue. do you want to keep this open? IMO having formatter with spdlog, https://github.com/ros2/rcl_logging/issues/92 would be the best place to discuss and closing this issue?

rr-tom-noble, 2024-09-03:
Hi @fujitatomoya. Yeah, that makes sense. Thanks for confirming

closed 2024-09-03 by rr-tom-noble

clalancette, 2024-09-03:
I think we have a resolution here (to move to a different issue), but I wanted to provide a couple of bits of additional context.

First, most of the details on why we moved from log4cxx to spdlog can be found in https://github.com/ros2/build_farmer/issues/189, https://github.com/ros2/rcl_logging/pull/78, and https://github.com/ros2/rcl_logging/pull/17.  In short, log4cxx had a lot of issues surrounding the lifetime of objects at the time I looked at this, and moving to spdlog solved all of them.

One other thing I'll mention is that, until https://github.com/ros2/rcl_logging/issues/92 is solved, you should be able to take a different (though less efficient path).  In particular, by default all logs go to the console, to disk, and to the network on the `/rosout` topic.  So if you really want to format your logs in a particular way, you can write a node that subscribes to `/rosout`, and writes them out however you want.  This does involve more network processing, but the good news is that this aggregates the logs from *all* nodes across the whole network.