ros2/rcutils #476: Whether the log time supported format as [2024-09-04 12:23:40]

congleetea, 2024-09-04:
Hi, The log time formats as timestamp is not good for find some information, maybe it formats as [2024-09-04 12:23:40] better?

fujitatomoya, 2024-09-04:
Please post questions like this to https://robotics.stackexchange.com/, where is our central QA site.

And you can do that with `RCUTILS_CONSOLE_OUTPUT_FORMAT` environmental variable, see more details for https://docs.ros.org/en/rolling/Concepts/Intermediate/About-Logging.html#environment-variables

```bash
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run demo_nodes_cpp talker
[INFO] [1725466479.904464075] [talker]: Publishing: 'Hello World: 1'
[INFO] [1725466480.904503836] [talker]: Publishing: 'Hello World: 2'
...
root@tomoyafujita:~/ros2_ws/colcon_ws# export RCUTILS_CONSOLE_OUTPUT_FORMAT="{date_time_with_ms} {name}: {message}"
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run demo_nodes_cpp talker
2024-09-04 09:15:43.213 talker: Publishing: 'Hello World: 1'
2024-09-04 09:15:44.213 talker: Publishing: 'Hello World: 2'
...
```

closed 2024-09-04 by fujitatomoya

congleetea, 2024-09-24:
thank you very much, it worked.