ros2/rcutils #477: rcutils_get_zero_initialized_string_array has (potential) data race

comicfans, 2024-09-04:
hello, during thread sanitizer test, I've found multi thread calling into rcutils_get_zero_initialized_string_array gives data race warnings from thread sanitizer

https://github.com/ros2/rcutils/blob/rolling/src/string_array.c#L35

```c
rcutils_string_array_t
rcutils_get_zero_initialized_string_array(void)
{
  static rcutils_string_array_t array = {
    .size = 0,
    .data = NULL,
  };
  array.allocator = rcutils_get_zero_initialized_allocator();  //.  <---- thread sanitizer report data race this line because caller on some other thread may read it, but every threading running this line is write to it, without synchronize
  return array;
}
```
this may behave correctly on some architect, but still a data race in C standard

clalancette, 2024-09-06:
Thanks for the report.  Looking at it, you are exactly right.

Can you give #478 a try with your tool to see if it fixes the problem for you?

comicfans, 2024-09-08:
confirmed this fix the tsan warning

PR #478 (2024-09-17): Fix setting allocator to NULL.
That is, previously the code was setting the allocator at runtime, not during static initialization.  Switch to static initialization, which should fix a possible race.

This should fix #477 .

A draft until we get confirmation from the reporter that it fixes the issue.
closed 2024-09-17 by clalancette