ros2/rcutils #482: Address sanitizer warns about mismatched new/delete types in string_map.c

skndi, 2024-11-19:
- ROS2 Humble
When running our application with AddressSanitizer it shows an error in string_map.c
`Nov 19 16:53:37 bash[5218]: ==5218==ERROR: AddressSanitizer: new-delete-type-mismatch on 0x007f77b34c10 in thread T2:
Nov 19 16:53:37 bash[5218]:   object passed to delete has wrong type:
Nov 19 16:53:37 bash[5218]:   size of the allocated type:   72 bytes;
Nov 19 16:53:37 bash[5218]:   size of the deallocated type: 1 bytes.
Nov 19 16:53:38 bash[5218]:     #0 0x7f7f7e910c in operator delete(void*, unsigned long) /development/libsanitizer/asan/asan_new_delete
Nov 19 16:53:38 bash[5218]:     #1 0x7f7c4fc85c in rcutils_string_map_fini (/opt/ros/lib/librcutils.so+0xc85c)
Nov 19 16:53:38  bash[5218]:     #2 0x7f7cfc6b04 in rcl_resolve_name (/opt/ros/lib/librcl.so+0x1bb04)
Nov 19 16:53:38 bash[5218]:     #3 0x7f7cfc6e60 in rcl_node_resolve_name (/opt/ros/lib/librcl.so+0x1be60)
Nov 19 16:53:38 bash[5218]:     #4 0x7f7cfc4ee8 in rcl_publisher_init (/opt/ros/lib/librcl.so+0x19ee8)
Nov 19 16:53:38 bash[5218]:     #5 0x7f7e77b9a4 in rclcpp::PublisherBase::PublisherBase(rclcpp::node_interfaces::NodeBaseInterface*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rosidl_message_type_sup
Nov 19 16:53:38 bash[5218]:     #6 0x7f7e6c79d8 in rclcpp::Publisher<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void> >::Publisher(rclcpp::node_interfaces::NodeBaseInterface*, std::__cxx11::basic_string<char, st
Nov 19 16:53:38 bash[5218]:     #7 0x7f7e6c7e24 in rclcpp::create_publisher_factory<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >, std::allocator<void>, rclcpp::Publisher<rcl_interfaces::msg::ParameterEvent_<std::allocator<void> >
Nov 19 16:53:38 bash[5218]:     #8 0x7f7e6c8258 in std::_Function_handler<std::shared_ptr<rclcpp::PublisherBase> (rclcpp::node_interfaces::NodeBaseInterface*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const
Nov 19 16:53:38 bash[5218]:     #9 0x7f7e6d07b4 in rclcpp::node_interfaces::NodeTopics::create_publisher(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::PublisherFactory const&, rclcpp::QoS const&
Nov 19 16:53:38 bash[5218]:     #10 0x7f7e6cae98 in rclcpp::node_interfaces::NodeParameters::NodeParameters(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeLoggingInterface>, std::shared
Nov 19 16:53:38 bash[5218]:     #11 0x7f7e68b104 in rclcpp::Node::Node(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const
Nov 19 16:53:38 bash[5218]:     #12 0x7f7e68d4f4 in rclcpp::Node::Node(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::NodeOptions const&) (/opt/ros/lib/librclcpp.so+0x2174f4)`

I tried looking into the rcutils source code, but it didn't seem there is any issue to me. Is this a false positive or is there a real problem? Thanks for the help in advance.

fujitatomoya, 2024-11-19:
duplication of https://github.com/ros2/rclcpp/issues/2220?

skndi, 2024-11-20:
Hi @fujitatomoya, that seems to be the case, yes. Should I close this one?

fujitatomoya, 2024-11-20:
yeah, let's close this one, we can track it with https://github.com/ros2/rclcpp/issues/2220