ros2/rcutils #486: compile error while using conan - humble

realysy, 2025-01-20:
I am trying to build ros2 humble from source on Ubuntu 20.04, and use [conan](https://docs.conan.io/2/integrations/ros.html#consuming-conan-packages-using-the-rosenv-generator) to try to fix some dependency problems of other packages. However, for package `rcutils`, I find that if I `source conan/conanrosenv.sh` before `colcon build`, `rcutils` will have compile error, even there is no dependency specified in `conanfile.txt`.

conanfile.txt: 
```ini
[requires]

[generators]
CMakeDeps
CMakeToolchain
ROSEnv
```

build.sh to call `colcon build`:
* Without `source conan/conanrosenv.sh`, `rcutils` can be built successfully.

```bash
source conan/conanrosenv.sh

export MAKEFLAGS="-j16"
export LLVM_INSTALL_DIR="/usr/lib/llvm-10"

colcon build --merge-install \
    --install-base ./humble \
    --parallel-workers 1 \
    --packages-select rcutils \
    --cmake-args -DCMAKE_BUILD_TYPE=Release
```

Compile error:
```log
Starting >>> rcutils
--- stderr: rcutils
In file included from /home/tom/ros2_humble/src/ros2/rcutils/test/./mocking_utils/filesystem.hpp:42,
                 from /home/tom/ros2_humble/src/ros2/rcutils/test/test_filesystem.cpp:25:
/home/tom/ros2_humble/src/ros2/rcutils/test/./mocking_utils/patch.hpp: In instantiation of ‘auto mocking_utils::make_patch(const string&, std::function<_Signature>) [with long unsigned int ID = 5; SignatureT = int(int, const char*, stat*) noexcept; std::string = std::__cxx11::basic_string<char>]’:
/home/tom/ros2_humble/src/ros2/rcutils/test/./mocking_utils/filesystem.hpp:163:3:   required from ‘class mocking_utils::filesystem::FileSystem<5>’
/home/tom/ros2_humble/src/ros2/rcutils/test/test_filesystem.cpp:219:30:   required from here
/home/tom/ros2_humble/src/ros2/rcutils/test/./mocking_utils/patch.hpp:294:71: error: ‘proxy’ has incomplete type
  294 | auto make_patch(const std::string & target, std::function<SignatureT> proxy)
      |                                             ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
In file included from /usr/include/c++/9/functional:59,
                 from /usr/include/c++/9/pstl/glue_algorithm_defs.h:13,
                 from /usr/include/c++/9/algorithm:71,
                 from /home/tom/ros2_humble/humble/src/gtest_vendor/include/gtest/internal/gtest-port.h:264,
                 from /home/tom/ros2_humble/humble/src/gtest_vendor/include/gtest/internal/gtest-internal.h:40,
                 from /home/tom/ros2_humble/humble/src/gtest_vendor/include/gtest/gtest.h:62,
                 from /home/tom/ros2_humble/src/ros2/rcutils/test/test_filesystem.cpp:15:
/usr/include/c++/9/bits/std_function.h:128:11: note: declaration of ‘class std::function<int(int, const char*, stat*) noexcept>’
  128 |     class function;
      |           ^~~~~~~~
In file included from /home/tom/ros2_humble/src/ros2/rcutils/test/./mocking_utils/filesystem.hpp:42,
                 from /home/tom/ros2_humble/src/ros2/rcutils/test/test_filesystem.cpp:25:
/home/tom/ros2_humble/src/ros2/rcutils/test/./mocking_utils/patch.hpp:296:10: error: invalid use of incomplete type ‘class mocking_utils::Patch<5, int(int, const char*, stat*) noexcept>’
  296 |   return Patch<ID, SignatureT>(target, proxy);
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/tom/ros2_humble/src/ros2/rcutils/test/./mocking_utils/patch.hpp:192:7: note: declaration of ‘class mocking_utils::Patch<5, int(int, const char*, stat*) noexcept>’
  192 | class Patch;
      |       ^~~~~
/home/tom/ros2_humble/src/ros2/rcutils/test/./mocking_utils/filesystem.hpp: In instantiation of ‘class mocking_utils::filesystem::FileSystem<5>’:
/home/tom/ros2_humble/src/ros2/rcutils/test/test_filesystem.cpp:219:30:   required from here
/home/tom/ros2_humble/src/ros2/rcutils/test/./mocking_utils/patch.hpp:316:66: error: parameter 2 of ‘auto mocking_utils::make_patch(const string&, std::function<_Signature>) [with long unsigned int ID = 5; SignatureT = int(int, const char*, stat*) noexcept; std::string = std::__cxx11::basic_string<char>]’ has incomplete type ‘std::function<int(int, const char*, stat*) noexcept>’
  316 |   decltype(mocking_utils::make_patch<id, decltype(function)>("", nullptr))
      |                                                                  ^~~~~~~
/home/tom/ros2_humble/src/ros2/rcutils/test/./mocking_utils/filesystem.hpp:163:3: note: in expansion of macro ‘MOCKING_UTILS_PATCH_TYPE’
  163 |   MOCKING_UTILS_PATCH_TYPE(ID, __xstat) __xstat_mock_;
      |   ^~~~~~~~~~~~~~~~~~~~~~~~
/home/tom/ros2_humble/src/ros2/rcutils/test/./mocking_utils/patch.hpp:294:71: note:   declared here
  294 | auto make_patch(const std::string & target, std::function<SignatureT> proxy)
      |                                             ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
...
[... 45 lines cut ...]
                 from /home/tom/ros2_humble/humble/src/gtest_vendor/include/gtest/internal/gtest-internal.h:40,
                 from /home/tom/ros2_humble/humble/src/gtest_vendor/include/gtest/gtest.h:62,
                 from /home/tom/ros2_humble/src/ros2/rcutils/test/test_filesystem.cpp:15:
/usr/include/c++/9/bits/std_function.h:667:7: error: ‘std::function<_Res(_ArgTypes ...)>::function(_Functor) [with _Functor = mocking_utils::filesystem::FileSystem<ID>::FileSystem(const string&) [with long unsigned int ID = 5; std::string = std::__cxx11::basic_string<char>]::<lambda(auto:1&& ...)>; <template-parameter-2-2> = void; <template-parameter-2-3> = void; _Res = __dirstream*; _ArgTypes = {const char*}]’, declared using local type ‘mocking_utils::filesystem::FileSystem<ID>::FileSystem(const string&) [with long unsigned int ID = 5; std::string = std::__cxx11::basic_string<char>]::<lambda(auto:1&& ...)>’, is used but never defined [-fpermissive]
  667 |       function<_Res(_ArgTypes...)>::
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/9/bits/std_function.h:667:7: error: ‘std::function<_Res(_ArgTypes ...)>::function(_Functor) [with _Functor = mocking_utils::filesystem::FileSystem<ID>::FileSystem(const string&) [with long unsigned int ID = 6; std::string = std::__cxx11::basic_string<char>]::<lambda(auto:1&& ...)>; <template-parameter-2-2> = void; <template-parameter-2-3> = void; _Res = __dirstream*; _ArgTypes = {const char*}]’, declared using local type ‘mocking_utils::filesystem::FileSystem<ID>::FileSystem(const string&) [with long unsigned int ID = 6; std::string = std::__cxx11::basic_string<char>]::<lambda(auto:1&& ...)>’, is used but never defined [-fpermissive]
/usr/include/c++/9/bits/std_function.h:667:7: error: ‘std::function<_Res(_ArgTypes ...)>::function(_Functor) [with _Functor = mocking_utils::filesystem::FileSystem<ID>::FileSystem(const string&) [with long unsigned int ID = 7; std::string = std::__cxx11::basic_string<char>]::<lambda(auto:1&& ...)>; <template-parameter-2-2> = void; <template-parameter-2-3> = void; _Res = __dirstream*; _ArgTypes = {const char*}]’, declared using local type ‘mocking_utils::filesystem::FileSystem<ID>::FileSystem(const string&) [with long unsigned int ID = 7; std::string = std::__cxx11::basic_string<char>]::<lambda(auto:1&& ...)>’, is used but never defined [-fpermissive]
/usr/include/c++/9/bits/std_function.h:667:7: error: ‘std::function<_Res(_ArgTypes ...)>::function(_Functor) [with _Functor = mocking_utils::filesystem::FileSystem<ID>::FileSystem(const string&) [with long unsigned int ID = 8; std::string = std::__cxx11::basic_string<char>]::<lambda(auto:1&& ...)>; <template-parameter-2-2> = void; <template-parameter-2-3> = void; _Res = __dirstream*; _ArgTypes = {const char*}]’, declared using local type ‘mocking_utils::filesystem::FileSystem<ID>::FileSystem(const string&) [with long unsigned int ID = 8; std::string = std::__cxx11::basic_string<char>]::<lambda(auto:1&& ...)>’, is used but never defined [-fpermissive]
/usr/include/c++/9/bits/std_function.h:667:7: error: ‘std::function<_Res(_ArgTypes ...)>::function(_Functor) [with _Functor = mocking_utils::filesystem::FileSystem<ID>::FileSystem(const string&) [with long unsigned int ID = 9; std::string = std::__cxx11::basic_string<char>]::<lambda(auto:1&& ...)>; <template-parameter-2-2> = void; <template-parameter-2-3> = void; _Res = __dirstream*; _ArgTypes = {const char*}]’, declared using local type ‘mocking_utils::filesystem::FileSystem<ID>::FileSystem(const string&) [with long unsigned int ID = 9; std::string = std::__cxx11::basic_string<char>]::<lambda(auto:1&& ...)>’, is used but never defined [-fpermissive]
[... cut ...]
make[1]: *** waiting for unfinished jobs....
In file included from /home/tom/ros2_humble/src/ros2/rcutils/test/test_time.cpp:26:
...
make[2]: *** [CMakeFiles/test_time.dir/build.make:76：CMakeFiles/test_time.dir/test/test_time.cpp.o] error 1
make[1]: *** [CMakeFiles/Makefile2:1061：CMakeFiles/test_time.dir/all] error 2
make: *** [Makefile:146：all] error 2
---
Failed   <<< rcutils [11.0s, exited with code 2]

Summary: 0 packages finished [12.0s]
  1 package failed: rcutils
  1 package had stderr output: rcutils
```

mjcarroll, 2025-01-21:
My initial guess would be this is from using an older version of LLVM.  Humble's tier 1 platform is Ubuntu 22.04, which ships with LLVM-14 by default.  

Can you see which compiler is used when you don't source the conan workspace and use that one?

clalancette, 2026-11-04:
No response in a long time, so I'll close this out.  If you are still having issues, please feel free to reopen.

closed 2026-11-04 by clalancette