ros2/rcutils #488: Use MONOTONIC_RAW instead of MONOTONIC for steady clock

saikishor, 2025-02-23:
I see that @clalancette recently switched to `MONOTONIC` from `MONOTONIC_RAW` in https://github.com/ros2/rcutils/pull/357 in order to have it similar to `std::chrono::steady_clock` representation. 

I've also seen the discussion in the issue: https://github.com/ros2/rcutils/issues/43. However, I think that we should align more towards the `MONOTONIC_RAW` because if I want to use the steady_clock provided by rcutils for the `ros2_control` applications, I want it to be slew-free rather than being affected by any internal clock change. For instance, Industrial grade motor control boards with EtherCAT protocol might generate some discontinuities in the velocities causing strange noises on the robot. 

It would be great to rethink again on this part to use `MONOTONIC_RAW` instead of `MONOTONIC` (or) if we can add new clock type something like `RCL_STEADY_SLEWFREE_TIME` or something like that to have both implementations?

Thank you

fujitatomoya, 2025-02-25:
IMO https://github.com/ros2/rcutils/pull/357 aligns with `std::chrono` to keep it similar with `std::chrono::steady_clock::now()` for `RCL_STEADY_TIME`.

> Industrial grade motor control boards with EtherCAT protocol might generate some discontinuities in the velocities causing strange noises on the robot.

this also sounds reasonable. sometimes we need to have pure hardware timer values without any adjustments by kernel, for example precise measurements.

i think these are different use cases depending on the application requirements. maybe we need to have a new clock type `RCL_STEADY_RAW_TIME`.

saikishor, 2025-02-25:
I understand that #357 aligns with the `std::chrono`. Indeed, it makes sense. It would be really great if we have a new clock type as you mentioned, this will be really useful for most robotics control applications.

Thank you

alsora, 2025-03-06:
Based on the discussion when this change was introduced, we think that we shouldn't revert it.
Adding a new type should allow to handle this use-case without affecting the existing users.

saikishor, 2025-03-06:
> Based on the discussion when this change was introduced, we think that we shouldn't revert it.
> Adding a new type should allow to handle this use-case without affecting the existing users.

Hello @alsora 

Thank you for responding. When I get sometime, I'll try to make the changes with the new type and open a PR.

fujitatomoya, 2025-07-12:
@saikishor just checking. did you or would you find some time to work on this?

PR #507 (2025-08-21): Add rcutils_raw_steady_time_now method for slew-free clock
fixes: #488 

This PR adds monotonic slew-free clock implementation that can be used for industrial applications where a slew-free clock plays a very important role when integrating ros2_control with industrial applications to avoid rattling sound in the actuators when there is a time drift
closed 2025-08-21 by fujitatomoya