ros2/rcutils #520: RcUtilsLogger does not implement logging.Logger

spacether, 2025-09-30:
### Generated by Generative AI

no

### Operating System:

```
<PUT OS INFO HERE>
```

### ROS version or commit hash:

humble

### 'ros2 doctor --report' output

<details><summary>ros2 doctor --report</summary>

```console
<COPY OUTPUT HERE>
```
</details>

### Steps to reproduce issue

Argument of type "RcutilsLogger" cannot be assigned to parameter "logger" of type "Logger" in function "load_camera_configs"
  "RcutilsLogger" is not assignable to "Logger"Pylance[reportArgumentType](https://github.com/microsoft/pylance-release/blob/main/docs/diagnostics/reportArgumentType.md)

### Expected behavior

I expect the RcutilsLogger to implement and extend logging.Logger, why does it not?

### Actual behavior

see above

spacether, 2025-10-03:
I am passing an instance of your logger and vscode reports that it is not an instance of logger.Logging. My ask is that you subclass that or implement it in your logger.

fujitatomoya, 2025-10-05:
can you provide the reproducible example code or package?